rsta2/circle #89: USB Multi-Touchscreen Driver Development

DC-INC, 2018-11-15:
Hello -

First, thanks for providing such a feature rich bare metal C++ environment for the PI.  After playing with a few examples I was eager to see about getting touch working for a ViewSonic TD2230.  I was never able to get multitouch support working under Raspbian for this monitor and using Circle seemed like a good starting point to see if I could get something to work.  In case this can help someone else I will document the steps it took to make this work.  Jump to the bottom to see questions related to current workarounds / known issues.

Initially the device was being configured as a gamepad since it was matching "int3-0-0" in usbdevicefactory.cpp during the USB enumeration process.  Obviously it was failing the configuration step since no axes, hats or buttons were in the report descriptor.  This stumped me for a bit since no warning was issued for failing the configuration step and it was returning FALSE on line 152 of usbgamepad.cpp.  Commenting out the return and allowing it to continue as a gamepad at least let it function as a pointing device.

At this point I could dump the device descriptor and see that indeed it could be configured as a multi-point touchscreen device.  Using the [USB.ORG : HID usage tables](https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf) I could at least start to understand what was being reported.

[image]

Since this monitor works as expected on Win7 and beyond I had a found the following useful:

1. [Using report descriptors to support capability discovery](https://docs.microsoft.com/en-us/windows-hardware/design/component-guidelines/using-report-descriptors-to-support-capability-discovery)

2. [Supporting usages in multitouch digitizer drivers](https://docs.microsoft.com/en-us/windows-hardware/design/component-guidelines/supporting-usages-in-multitouch-digitizer-drivers)

3. [Single finger hybrid mode report descriptor](https://docs.microsoft.com/en-us/windows-hardware/design/component-guidelines/single-finger-hybrid-mode-report-descriptor)

Sure enough, the device reports a configuration mode as shown in the first Microsoft document.  Using `CUSBGamePadDevice::PS3Configure` as a reference and setting device mode (0x52) to multi-touch (0x02) based on the same MS document allowed the device to start sending report id 1.

```
void CUSBMultiTouchDevice::VendorConfigure (void)
{
	enum
	{
		Mouse = 0x00,
		SingleInput = 0x01,
		MultipleInput = 0x02,
	};

	static u8 DeviceMode[] = {0x52, MultipleInput};
	int result = GetHost ()->ControlMessage (GetEndpoint0 (),
					    REQUEST_OUT | REQUEST_CLASS | REQUEST_TO_INTERFACE,
					    SET_REPORT, (REPORT_TYPE_FEATURE << 8) | 0x43,
					    GetInterfaceNumber (), DeviceMode, sizeof DeviceMode);

	if (result > 0) m_usReportSize = 108;
}
```

This is where I had to abandon using `CUSBGamePadDevice::DecodeReport`.  The logic around which report id to use was causing issues as well as the initial configuration step to determine the report length usbgamepad.cpp line 144 -

```
u8 ReportBuffer[8] = {0};
DecodeReport (ReportBuffer);
```

Based on the report descriptor I calculated the report length should be less than 128 bytes so I hardcoded this into `CUSBHIDDevice::Configure(128)` and added logging to `CUSBHIDDevice::CompletionRoutine` to display report length and payload -

[image]

At this point I copied usbgamepad.h/cpp to usbmultitouch.h/cpp and backed out my changes to usbgamepad.  Then I updated the USB device factory to use the corresponding vendor / product id pair and treat this as a usb multitouch device - 

[image]

**Current State / Workarounds / Questions**

Current workarounds are related to being notified when the last touch point is released.  The monitor does not respond to queries (NAK) when touch points are not available and the driver is not notified of the NAKs.  The first though was to enable `CUSBRequest::SetCompleteOnNAK` and though this does notify the driver (after removing an assert for not being a bulk transfer) the performance is horrible on core 0.  My guess is this is caused by restarting the kernel timer, but I did not investigate further.  Basically I added a `CUSBRequest::SetErrorRoutine` which is called from `CDWHCITransferStageData::TransactionComplete` -

```
        if (nStatus & DWHCI_HOST_CHAN_INT_NAK)
        {
            m_pURB->CallErrorRoutine ();
        }
```

I may be missing something obvious, but is there a better way to get notified of the NAK?

Sorry for the long winded description, but from seeing other questions related to missing drivers, etc. I figured this could help someone out down the line.  Once the driver is cleaned up a little more I can push this for inclusion into the main trunk.  Mainly, the parsing is very rigid and not like the generic gamepad which loops over the report descriptor.  In theory this should work similar once the correct report id is discovered and the report length is calculated instead of being hardcoded.

Thanks again for providing the Circle core as it really is an excellent bare metal framework.

Best Regards

rsta2, 2018-11-16:
Thanks for your words about Circle. Your initiative to develop a driver for the ViewSonic TD2230 touchscreen and to document your work is much appreciated, because drivers are very important and unfortunately I cannot do this all on my own, because often I do not have the hardware.

Some notes:

The "int3-0-0" interface is currently dedicated to HID standard gamepad devices only. That's the reason why the driver returns FALSE from Configure(), if there are no gamepad controls found in the HID report descriptor. For instance USB keyboards or mice have this interface too, but are not controlled using it. They have another "int3-1-X" (X = 1, 2) interface, which is used here. It's the intended way to detect devices like the TD2230 using vendor and product ID.

It's a common way now, not to use the HID report descriptor parser in DecodeReport() and to hard-code the report length and report structure. We are currently working on new gamepad drivers and we are doing the same there. The HID report descriptor parser is not complete and it would be more difficult to extend it, but to hard-code these things.

About the NAK problem:

I guess, this touchscreen is a high-speed USB device. I would need to know, if it is. You can remove the comment in line 215 in the file *lib/usb/usbstandardhub.cpp* to get the port status of this device. If bit 10 of the reported value is set, then the device is high-speed. This is important, because your touchscreen would be the first USB device, which is working with an interrupt-in endpoint with high-speed and this is the reason, why *CUSBRequest::SetCompleteOnNAK()* is not working. I have to update the USB stack for it. After that you should be able to use *CUSBRequest::SetCompleteOnNAK()* and an USB request will complete with *GetStatus()* != 0 (success) and *GetResultLength()* == 0 if NAK has been received. This should be a clean solution.

I'm interested in your driver to be included into the Circle USB library. If my assumptions above are correct, I will add a feature branch "touchscreen" and will make the proposed USB update there, so you can test it with your driver. It would be good, if we can establish a common interface for touchscreens, so that an application does not need to know, which concrete hardware it is using. Perhaps you know, that there is already a driver for the Official 7" Raspberry Pi touchscreen in Circle. If its interface does not fit for the TD2230, we should make it fit.

DC-INC, 2018-11-16:
I don't believe it is high-speed as bit 10 seems to be cleared after the device is configured.  The monitor reports as _ven2575-401_ -

[image]

Let me describe what I am seeing when I call _CUSBRequest::SetCompleteOnNAK()_.  I started with the multi-core fractal example and removed the fractal drawing code.  Only the temperature task is running on core 0 along with a loop that does nothing but increment a counter and display the count and clock ticks.  The other 3 cores are also incrementing a counter and displaying clock ticks.  Cores 1-3 increment at the same rate and core 0 has a slight performance hit since the temperature task running at 50 Hz (_MsSleep(20)_).  After 100,000 counts it would look something like -

```
Core	Counts
 0	95,000
 1	100,050
 2	99,999
 3	100,025
```

The _CUSBRequest::SetCompleteOnNAK()_ is hidden inside _CUSBHIDDevice::StartRequest()_ or at least that is where it appeared this should be set.  Obviously setting this here affects all HID devices and there should be a flag if this is required, but this is what I currently have (replace _SetErrorRoutine_ with  _SetCompleteOnNAK_) -

```
boolean CUSBHIDDevice::StartRequest (void)
{
	assert (m_pReportEndpoint != 0);
	assert (m_pReportBuffer != 0);
	
	assert (m_pURB == 0);
	assert (m_nReportSize > 0);
	m_pURB = new CUSBRequest (m_pReportEndpoint, m_pReportBuffer, m_nReportSize);
	assert (m_pURB != 0);
	m_pURB->SetErrorRoutine (ErrorStub, 0, this);
	m_pURB->SetCompletionRoutine (CompletionStub, 0, this);

	return GetHost ()->SubmitAsyncRequest (m_pURB);
}
```

The device is notified of the completion, but core 0 is incrementing the counter at ~ 100 HZ.  After 100,000 counts the results are like -

```
Core	Counts
 0	1,000
 1	100,050
 2	99,999
 3	100,025
```

Basically a 100x reduction in throughput on core 0.  Restarting the transaction for a NAK seems like a waste of resources when a _virtual_ stub for errors seemed like a better fit  -

```
virtual void CUSBHIDDevice::ErrorHandler (u32 nErrorCode) { /* nop for most device types */ }
```

Hope this helps to diagnose what I am seeing here.  Let me know what changes to make and I can send screenshots of how core 0 performs.

The usb touchscreen should easily work within the existing framework once _CTouchScreenDevice_ is reworked to use an input provider.  Define an input interface which allows _CTouchScreenDevice_ to query for touch points and related id, xpos, ypos and then any touch type device can be used.

Any thought to making each input type a task and having them run at what ever frequency is required?  From my limited exposure to the Circle API, core 0 is the I/O core and main UI core so there should be plenty of spare cycles available and removes the need to have a separate for (;;) loop in the _Kernel::Run_.  Just define an idle task that does nothing but call _Yield()_.  This assumes of course you are always running multi-core and can move the application logic to another core, but even single-core mode this still should work seamlessly.

Actually, defining an idle task would be the start of a task manager like interface showing CPU usage.  Have _Yield()_ increment a counter and reset every second and base the counts on a calibrated number of known counts/sec.  At least for me, I could see wrapping the temperature monitor and this together as a sort of overall performance monitor.

rsta2, 2018-11-16:
First, your touchscreen is a full speed device behind a hub. That means it uses USB split transfers. 

I think, I know what is causing this bad performance on core 0. If you set *SetCompleteOnNAK()* in *CUSBHIDDevice::StartRequest()* any USB request will be terminated immediately. It does not play a role, if data from the touchscreen is available. The request is completed immediately. And will be restarted in *CUSBHIDDevice::CompletionRoutine()* again and again. That means the core does nearly nothing else but polling the touchscreen interrupt endpoint.

Your touchscreen is a special device, because the driver needs the information that an USB request has been answered by a NAK. This is not the case for normal USB HID devices. That's why your driver class cannot be derived from the class *CUSBHIDDevice*. You have to include the function of *CUSBHIDDevice* into your driver class. It will probably have a *StartRequest()* method too and you should set *SetCompleteOnNAK()* there.

To dramatically reduce the core 0 load you have to start a kernel timer, when a request is completed with *GetStatus()* != 0 and *GetResultLength()* == 0 to have some delay between two polls to the device, which are answered with NAK. This is already implemented in an equivalent way in *usbmidi.cpp* for MIDI devices [here](https://github.com/rsta2/circle/blob/master/lib/usb/usbmidi.cpp#L217). Perhaps you can take this as example. I will have to change the `assert (m_pEndpoint->GetType () == EndpointTypeBulk);` in *dwhcixferstagedata.cpp* so that *EndpointTypeInterrupt* is allowed too.

I have already an idea, how a touchscreen interface class can be implemented. Basically there will be a new class *CTouchScreenFT5406Device*, which implements the current function of *CTouchScreenDevice*. *CTouchScreenDevice* will be the new interface class. I would implement this, when your driver is working without these load problems. Your driver class may be named *CUSBTouchScreenTD2230Device*.

Please note that the scheduler in Circle is optional. It is not documented somewhere, but the USB library must not rely on the scheduler as it is the case for the input library and the FAT filesystem library. That means you cannot use the scheduler to implement your touchscreen driver (if it should become a part of Circle) or the touchscreen interface. There are many sample programs, which do not use the scheduler and this will remain the case, because it is one strength of Circle, that it can be used to develop both smaller and larger applications.

DC-INC, 2018-11-16:
Thanks for the USB overview.  I had investigated just enough to realize I was getting deeper and deeper in the weeds and may not be able to find my way out <grin>.  It was the `assert (m_pEndpoint->GetType () == EndpointTypeBulk);` that got me at first and then when this was commented out that I saw the endless `CUSBHIDDevice::StartRequest()` / `CUSBHIDDevice::CompletionRoutine()` dance.

As I eluded to earlier, having the NAK not complete but at least notify the device may be useful in the future and would be a _nop_ for existing HID devices.  Maybe I am missing something in the bigger picture, but it is a one line change to `CDWHCITransferStageData::TransactionComplete`.  I can work with it either way, but it seems cleaner to have the USB stack notify of NAK (or other error code and possibly take corrective action) than to schedule a kernel timer.

As for the input task, I was thinking at a higher level and nothing interrupt based.  Basically the classes inside '/lib/input/': keyboardbehaviour.cpp, mousebehaviour.cpp and touchscreen.cpp.  Anything that requires polling:

```
for (;;)
{
    pDevice->Update(); // or similar UpdateXXX() logic
    MsSleep(20); // 50 HZ
}
```
As the mouse & keyboard are implemented, the events appear to be generated at the interrupt level and and continue to bubble up through the layers of event handler callbacks.  I was thinking of having an event queue that is populated at _IRQ_LEVEL_ and the queue is emptied at _TASK_LEVEL_.  This avoids many deadlock scenarios that creep up as more and more devices / logic are added to a running system.

I agree keeping it as simple as possible is the best path forward.  It is just as easy to wrap what I was proposing at the _RegisterEventHandler_ level with a class that implements the event interface, event queue and task all in one.  Sometimes you just have to look at it from another angle and your comment made that happen.

I will play around with a quick prototype that wraps the mouse / keyboard / touchscreen behaviors and implements simple tasks for broadcasting to the respective listeners at `TASK_LEVEL`.  I want to separate the interrupt notifications / updates as much as possible from task logic.  Please correct me if my understanding of the device driver model is incorrect.

Regards.

DC-INC, 2018-11-17:
A couple of things came into focus as I wrote the behavior.  The most important is that you don't need the _NAK_ acknowledgement as either a timer can be added to the behavior or if task based then when the queue empties you know the last point(s) are no longer pressed and can be cleared.  The initial version was tasked based and I will refactor this into a non-task version shortly.

Both methods at least remove any required modifications to the USB stack.

rsta2, 2018-11-17:
The device driver model for example for the USB HID devices is simple: Request, interrupt, request restarted after delay (on NAK) or callback.

Believe me, I understand what your problem is and I have been thinking about adding a work queue or extending the range of the scheduler more than once. But this is not only a technical decision. I think, Circle lives in a niche between all these RTOSs with sophisticated concepts and schedulers, which are not easy to understand. Circle is different and should be easy to understand, even without a big documentation, at least for application developers.

Writing drivers for Circle may be not that much fun like for other OSs, where you easily can start a kernel thread to do things, which have to be done regularly. But only a few people are doing this, more people are writing applications. I think, most of the applications have a main loop and it's not difficult to call some update function from it, which needs to be on TASK_LEVEL. Changing this would be a big decision, which would require much more gain from it.

rsta2, 2018-11-25:
Are you still working on the touchscreen driver or have you been deterred by my last comment? That would be a pity. I'm still interested in your driver. You wrote, you managed it, to get it working without changes to the Circle USB stack. That's good and should make it easier to integrate it. I'm not against modifying Circle, but modifications have to be moderate for a new driver. There are a lot of dependencies in Circle and with other projects, which have to be considered.

DC-INC, 2018-11-26:
Sorry, we have been away for the last week (Thanksgiving Holiday) and just getting back home. Yes, I am still developing the driver, but got side tracked with getting a network based file uploader working based on the web server sample.  The serial uploader doesn't work for me because the serial ports being used and swapping cables is just as painful as the SD card dance.  At some point I will make _enctype='multipart/form-data'_ work and it will be both faster and support file names.  For now I just needed a simple way to get new builds sent to the device without having to do anything and sample 21 was the easiest path.  It will be a few days before I get back to this project as I have to catch up on being away for a week.

Also, your last comments did not deter me.  Like I said in a previous comment, I can easily work around this and have a prototype working based on how I want it to work.

Jakar510, 2019-08-19:
Any update on the status of the driver? I'm using the TD2230 monitor and the Raspberry Pi 3b+/4b doesn't reliably catch all touch events (some are missed/lost).

rsta2, 2019-08-21:
There is currently no TD2230 touchscreen driver in Circle. Unfortunately I do not know the status of the driver of @DC-INC.

rsta2, 2019-09-04:
How it seems, this driver development has not been continued. Therefore the issue is closed.

closed 2019-09-04 by rsta2