rsta2/circle #91: Support for A64 on RPi 3?

winstyfreddy, 2018-11-28:
Hi Rene,

I'm really happy to find this repository of yours, and really appreciate all the hard work you've done to create it.

I can successfully build and run samples, but did not see a way to generate an aarch64 kernel for RPi 3B+. I guess this could be done by making local changes to the `ARCH` flags in top level `Rules.mk`, but maybe this would be nice to have an alternative value for `RASPPI` to enable this, such as `3-A64` rather than `3`. What do you think? Or maybe a different (orthogonal) property for setting the instruction set, which defaults to A32, but can be set to Thumb, A32 or A64 by overriding it?

I'm really excited about using your library for my bare metal project, many thanks again for all your hard work (especially implementing the USB 2.0 standard!).

petemoore, 2018-11-28:
Sorry, I've just discovered https://github.com/rsta2/circle64 - so I think this issue can be closed. Sorry, I had not seen that before!

I don't see a way for me to close this github issue myself, I don't think I have the authority to do so, sorry! But consider it solved, as I see the other repo.

Maybe the only thing that could be done is linking to one project from the other, to highlight that there are two versions. That might be worthwhile in case somebody else makes the same mistake as I did. :-)

winstyfreddy, 2018-11-28:
Sorry, I've just discovered https://github.com/rsta2/circle64 - so I'll close this!

Note, it might be worth linking from one repo to the other, in case somebody else makes the same mistake.

closed 2018-11-28 by winstyfreddy