rsta2/circle #92: FatFS : seeking and writing only part of a file

stephanedamo, 2018-12-10:
Hello,

For my project I'd need to write only at some locations in a file. It could be done by reading the whole file into memory, doing the modifications and re-writing everything but it has too much drawbacks for my use (slow, more wear to the flash card).

I was thinking about the classic C-way of doing it, opening a file in R+ mode, but it seems it's not possible to do that with the current CFatFileSystem class, which only allows to write a file from scratch and has no seek method. Is it feasible to modify it so it can handle that ? I've tried to do some modifications to your code, but I'm probably missing something since the opened file is written but then the FAT of the next file is corrupted.

rsta2, 2018-12-11:
The Circle-native file system support is limited and does not support seeking in a file. But there is an alternative. Have a look at the ported [FatFs](https://github.com/rsta2/circle/tree/master/addon/fatfs) file system module in addon/. It should support everything you want.

rsta2, 2018-12-11:
No problem! It's good, if it helps.

stephanedamo, 2018-12-12:
I have a lot of crashes since I replaced your CFatFileSystem with FatFs.
I noticed I had a freeze and sometimes an assert If i tried to read too much data in one go (like more than 1Kb)

> emmc.cpp  line 900 : assert (((u32) m_buf & 3) == 0);

After splitting the reads into smaller ones, even if some files are correctly read, some others aren't and the program is now really unstable. Freezes / undefined instructions errors/....

I'm using it like in your example:  
* initialize logger, timer, interrupt and emmc (in this order)
* call f_mount
* reading some files

 In my real program I'm also using I2S, I2C, serial (with FIQ) and display at the same time. Is it possible that there is a conflict somewhere ? (CFatFileSystem being slower because performing more checks to ensure reliability ?)

rsta2, 2018-12-12:
First, CFATFileSystem has been developed by myself. It should be stable but slow and has limited function only. FatFs is a 3rd party product and has been adapted to Circle, which is good documented. Unfortunately I do not know the internals of FatFs. The link to the FatFs website is in the sample README. It is not easy and very time consuming to develop a stable file system driver with good performance. The EMMC driver for the physical SD card access is also 3rd party and has been ported to Circle with a lot of modifications.

Before we go further into searching for the problem, I think, the initialization order in your program is wrong. The right order is: logger, interrupt, timer, emmc. See [here](https://github.com/rsta2/circle/blob/master/addon/fatfs/sample/kernel.cpp#L58). This order is very important, because the timer uses the interrupt object. Can you please correct this and check, if the problem remains.

stephanedamo, 2018-12-12:
Hello,
Thanks for your help, I fixed the initialization order but it didn't fixed the problem.

I found a way to reproduce it : 
* Create a file of at least 1024 bytes, name it "test.tes" and put it on the root of the SD card
* Compile the following code and put it on the SD card : [main.zip](https://github.com/rsta2/circle/files/2673417/main.zip)
* Run the program, it should give the following error : 
[image]

How does this crash ? I don't know, but it all depends on the sizes I try to read from the file : 
```
// CRASH !
f_read (&file, tempData, 1, &nBytesRead);
f_read (&file, tempData, 1500, &nBytesRead);
```
```
// Does not crash
f_read (&file, tempData, 1500, &nBytesRead);
f_read (&file, tempData, 1500, &nBytesRead);
```
```
// Does not crash
f_read (&file, tempData, 1, &nBytesRead);
f_read (&file, tempData, 2, &nBytesRead);
f_read (&file, tempData, 3, &nBytesRead);
f_read (&file, tempData, 4, &nBytesRead);
f_read (&file, tempData, 5, &nBytesRead);
f_read (&file, tempData, 6, &nBytesRead);
```

Doing f_read() past the file's size has no impact (probably handled correctly by FatFS). Crash doesn't seems to happen with files smaller than 1024 bytes (1023 seems fine). When the file is bigger than this limit, it may or may not crash depending on the read patterns... but it's reproductible, in my tests example #1 always crashes, examples 2 and 3 never does.

rsta2, 2018-12-13:
I checked this. As you already noticed in your last comment, the FatFs module provides an unaligned buffer in some circumstances. This should not be a problem for the EMMC driver, because it uses programmed I/O. **The assert() in line 900 can simply be removed for the moment.** This assert() was there, because the EMMC driver worked with DMA formerly and this support has been removed. But I have to investigate this further, because FatFs can also be used with USB drives and the USB mass-storage driver requires aligned buffers. You should update your Circle copy with `git pull`, because there is a new release 37 out. Please let me know, if there are further problems (after removing the assert()).

rsta2, 2018-12-13:
I implemented a solution for the problem of unaligned disk accesses in FatFs. It's currently on the *develop* branch and needs further testing. With this update the assert() in the EMMC driver does not need to be removed any more and it should work with USB drives too.

stephanedamo, 2018-12-13:
It works like a charm, thank you very much for the time you spent fixing this !

For my devs I've added to Circle the ability to control the backlight brightness - if you're interested I can do a pull request.

rsta2, 2018-12-14:
Thank you for reporting this and for testing!

Yes, please do a pull request. Please branch off from the *develop* branch.

closed 2018-12-14 by stephanedamo