rsta2/circle #93: support for wifi

Siddhartha123, 2019-01-09:
Is it possible to implement wifi AP in circle ?

Siddhartha123, 2019-01-11:
So there is no way to transfer data wirelessly in a bare metal environment ?

rsta2, 2019-01-11:
I have read about solutions using the external ESP8266 chip, which provides WiFi support via a serial AT command interface.

Siddhartha123, 2019-01-11:
That is definitely one way. But I would like to use the onboard wifi controller of Rpi 3.

rsta2, 2019-01-11:
Unfortunately I do not know of any bare metal solution for the onboard controller of the RPi 3. It's difficult.

vanvught, 2019-06-16:
The firmware for the BROADCOM Wifi chip is a blob. Hence it would be (almost) impossible to support in baremetal. Therefore,  I have designed a 4-bit parallel ESP8226 interface -> https://github.com/vanvught/rpidmx512/blob/master/lib-esp8266/src/rpi/esp8266.c

JeanValjean2, 2019-06-17:
Out of curiosity, could you point me to the blob from Broadcom ? I completely overlook the wifi part of the board...

vanvught, 2019-06-17:
@JeanValjean2 You would need to check the Linux driver(s)

Siddhartha123, 2019-06-23:
> Out of curiosity, could you point me to the blob from Broadcom ? I completely overlook the wifi part of the board...

check this out- 
https://www.raspberrypi.org/forums/viewtopic.php?t=203508

JeanValjean2, 2019-06-24:
Thanks to both of you, I'm now reading avidly the Cypress documentation :)

rsta2, 2020-03-14:
There is a development version of Wi-Fi support on the *wifi* branch. The Wi-Fi driver should work on all Raspberry Pi models with on-board Wi-Fi device and can be found in [addon/wifi/](https://github.com/rsta2/circle/tree/wifi/addon/wifi). It has been ported from the Plan 9 system.

There is a sample program, which demonstrates Wi-Fi usage in the sample subdirectory. Please read the [README](https://github.com/rsta2/circle/blob/wifi/addon/wifi/sample/README) for details! This version is in an early state and has several known issues. Not all features of the driver can be used currently.

Please note that the sample program will not work on the RPi 3B and 3B+ without the following patch applied (on RPi 4B and Zero W it should work without). This is caused by the different numbering of the network devices.

```
diff --git a/lib/net/netdevlayer.cpp b/lib/net/netdevlayer.cpp
index 4a53b85..b27891b 100644
--- a/lib/net/netdevlayer.cpp
+++ b/lib/net/netdevlayer.cpp
@@ -48,7 +48,7 @@ boolean CNetDeviceLayer::Initialize (boolean bWaitForActivate)
 #endif
 
 	assert (m_pDevice == 0);
-	m_pDevice = CNetDevice::GetNetDevice (0);	// get "eth0"
+	m_pDevice = CNetDevice::GetNetDevice (1);	// get "eth1"
 	if (m_pDevice == 0)
 	{
 		CLogger::Get ()->Write (FromNetDev, LogError, "Net device not available");
```

Currently the ported driver works with open Wi-Fi networks (without authentication) only.

Siddhartha123, 2020-03-16:
This is great. I will surely try this out. Thanks for the addon.
As this is a great starting place for bare-metal wifi support, I am closing this issue.

rsta2, 2020-03-21:
Thanks. The *wifi* branch has been updated and supports WPA2 enabled networks now. I have ported *WPA Supplicant* to make this possible. It is included as a submodule in *addon/wifi/hostap*. The patch for the RPi 3 is not needed any more. I reopen this issue for further info and maybe discussion on Wi-Fi support in Circle.

hpingel, 2020-04-10:
Hi Rene, Thank you for this wifi branch. I played with it a bit and (still being new to circle) have a question and also want to point out that I started a build with define ARM_ALLOW_MULTI_CORE set (by accident) and saw that the build fails with this being active:
```
In file included from p9proc.cpp:1:
p9proc.cpp: In function 'void __p9lock(Lock*)':
p9proc.h:20:14: error: 'Lock' {aka 'struct lock_t'} has no member named '__p9lock'
   20 | #define lock __p9lock
      |              ^~~~~~~~
p9proc.cpp:47:29: note: in expansion of macro 'lock'
   47 |   : : "r" ((uintptr) &lock->lock) : "x1", "x2", "x3"
      |                             ^~~~
p9proc.cpp: In function 'void __p9unlock(Lock*)':
p9proc.h:20:14: error: 'Lock' {aka 'struct lock_t'} has no member named '__p9lock'
   20 | #define lock __p9lock
      |              ^~~~~~~~
p9proc.cpp:76:29: note: in expansion of macro 'lock'
   76 |   : : "r" ((uintptr) &lock->lock) : "x1"
      |                             ^~~~
../../Rules.mk:149: recipe for target 'p9proc.o' failed
make: *** [p9proc.o] Error 1

```
Next to that I would like to ask if there is a specific reason why the firmware and wpasupplicant.conf are being read from a USB device. I tried to change the code a bit to be able to read the files from SD card, but I'm not too familiar at the moment with EMMC/FATFS filesystem mounting. Do you think it could be possible for me to change the code to SD access?

Thanks in advance,
Henning

rsta2, 2020-04-10:
@hpingel There was a name clash with "lock" and ARM_ALLOW_MULTI_CORE. It should be fixed now. Thanks for reporting this issue!

The WiFi device and SD card driver are using the same EMMC interface device at the moment. That's why they cannot be used at the same time. Perhaps there will be a driver for the SDHOST interface of RPi 1-3 and the EMMC2 device of the RPi 4 later, so that the firmware and configuration can be read from the SD card too, but this will take time.

hpingel, 2020-04-10:
> The WiFi device and SD card driver are using the same EMMC interface device at the moment. That's why they cannot be used at the same time.

Ah, I see. Thank you for explaining this.

rsta2, 2020-04-11:
The WLAN support is now on the *develop* branch to be included in the next release. The related source code is now in the subdirectory *addon/wlan/* and will remain "Experimental" in the release, because there is not enough feedback, how good it works. I renamed some things, because I don't want to touch registered trademarks here.

mhaddon, 2020-04-11:
> The WiFi device and SD card driver are using the same EMMC interface device at the moment. That's why they cannot be used at the same time.

Hey rsta2, I have been playing around with this wifi branch a bit and it is great, the only downside is this issue. It is actually the main reason why I cannot use this feature release.

Is this something that will be resolved in a future release? Is there anything we can do to help its implementation?

rsta2, 2020-04-12:
@mhaddon Thanks for your feedback and offer to help! What we need are two drivers, one for the SDHOST controller of the RPi 1-3 and Zero and one for the EMMC2 controller of the RPi 4. Both controllers can be used in parallel to the on-board WLAN controllers to access the SD card on the respective RPi models.

I already ported the Linux SDHOST driver to Circle to some degree, but it has major issues:

* It does currently not work with ARM_ALLOW_MULTI_CORE enabled on RPi 2-3. In the way it is implemented, a spin lock is acquired recursively, which is not supported by Circle. The structure of the driver is a bit complicated, so that this cannot be solved easily.
* It does programmed I/O from the IRQ handler to read/write data blocks, which takes time and increases the IRQ latency very much. This is not acceptable in a system like Circle.

I did some experiments to use the existing SD card driver in Circle with the EMMC2 device of the RPi 4, but without success. I'm missing some information here, but a Linux driver is available and could be analyzed.

So it is planned to solve this USB-only issue in the future, but I can only go step by step.

I don't know, which RPi model you want to use by yourself? If it is a RPi Zero W, perhaps the ported SDHOST driver may help you already. I could push it to a new branch in the Circle repo. The same is the case, if you do not need ARM_ALLOW_MULTI_CORE on a RPi 3. I can also give more information, what I tried with the EMMC2 controller of the RPi 4.

rsta2, 2020-04-24:
The driver for the EMMC2 controller of the RPi 4 is done on the *develop* branch. So parallel access to a WLAN and SD card is possible now on RPi 4. This is not implemented in the WLAN sample yet, but is not difficult. Improving the SDHOST driver for RPi 1-3 and Zero requires more effort. Perhaps I have to implement this driver again from scratch.

rsta2, 2020-05-01:
The SDHOST driver for RPi 1-3 and Zero is finished for the moment. It's still not perfect, but should work. So parallel access to WLAN and SD card on RPi 3 and Zero W is possible too now. I will work towards preparing the release Step42 now.

hpingel, 2020-05-03:
Hi Rene,

I was playing with you current develop branch to hack WLAN support into a circle application just to see how far I would get. Currently when I try to mount the filesystem from SD card on my Raspberry Pi 3 B+ I run into this error once the WLAN addon tries to read the firmware file (which is present on the SD card):
`emmc: EnsureDataMode() error sending CMD13`

It could just be something small I impatiently overlooked when adding the WLAN stuff to the code.

rsta2, 2020-05-03:
I guess, you have defined the *REALTIME* option in *include/circle/sysconfig.h*? *USE_SDHOST* is normally defined automatically on the RPi 3, but not if *REALTIME* is defined. Be sure to rebuild all libraries after modifying a system option, especially in *Config.mk*!

USE_SDHOST is not needed and not supported on the RPi 4 and is normally disabled there. The RPi 4 has a second EMMC controller, which is used for interfacing the SD card.

hpingel, 2020-05-03:
Thank you for your quick response! You were right: I am using REALTIME and I didn't make fully clean on makeall clean (WLAN addon wasn't cleaned properly). I removed the REALTIME option for the moment and it is running on the Rasppi 3 aarch 64! :-)

Doing further testing and playing now...

hpingel, 2020-05-06:
Hi Rene, When testing enabled WLAN (with the Sidekick64 - formerly known as RasPiC64 application) I noticed that there is a behaviour that an active WLAN connection seems to time out or disable itself after a certain time - if I do a keypress after three minutes of doing nothing and that keypress triggers a new HTTP GET request, the WLAN tries to re-establish the connection again including the authentication. Is this an intentional behaviour or could this be related to the specific settings of Sidekick64 (Realtime/FIQ). Or is there a way to configure this timeout?

rsta2, 2020-05-06:
My guess is, that a disconnect can occur in a WLAN and if the connection is re-established automatically, when there is some data to be transferred, then it should be OK. I'm not aware of a configurable timeout. But I am not a WLAN expert. This is one reason, why the WLAN support is noted to be "experimental".

rsta2, 2020-05-09:
Circle Step42 with experimental WLAN support has been released.

closed 2020-06-16 by rsta2

rsta2, 2021-11-02:
@hpingel There are some news on the connection breaks, you have discovered. I'm using the WLAN support in Circle for a local application 24 hours per day and I noticed such breaks too. But it depends on the used access point, if this happens. With one AP I do not have any breaks at all, but with an other AP the connection timeouts after every 5 minutes without data transfer. From the things I did read on the Internet, I suppose this timeout is a power saving feature and it depends of the AP model, if it is implemented and is configurable. Did you have more findings in this field? I would like to promote the Circle WLAN support from experimental, because my impression is that it is stable now.

hpingel, 2021-11-02:
@rsta2: Thank you for your question.

In the meantime I have changed my coding a bit and this has improved the situation. What I had ignored the last time is that scheduler->yield() has to be called regularly to make the WLAN task happy. In the context of Sidekick64 the specific situation is that it constantly uses the FIQ handler to deal with the GPIOs and normal interrupts can only be enabled for a very short time inbetween. So in my code to deal with network I am always disabling the FIQ handler, doing network stuff and then re-enabling the FIQ handler again. At first I only put in one single scheduler->yield() in the network "phase" but I then realized that I can afford to call scheduler->yield() many times and this does "good" to my webserver task and also to the WLAN task. At the moment (as a rule of thumb) I call scheduler->yield() one hundred times in a loop during the network phase... This is very arbitrary but just improves WLAN and webserver traffic a great deal.

While Sidekick64 is doing some emulation modes that have to be even more precise I can't afford to squeeze network "phases" into that timing-wise. For those situations I would like to have the opportunity to disable a Circle task so that if scheduler->yield() is called for other purposes then network stuff I don't want the kernel to spend cycles on the network stuff.... 

I believe that in Circle apps like the excellent mt32-pi (@dwinham) where the GPIO handling may be more "relaxed" timing-wise there should be no problem with WLAN. Having said that while I was testing mt32-pi recently I was not able to successfully use this rtp-MIDI feature yet.

hpingel, 2021-11-02:
Oh, I forgot another thing: I would like to check in my code if WLAN became inactive to react to it properly. What I want is to show a message on the "screen" (Commodore 64 screen) to inform the user that a WLAN reconnect is being attempted. For this I tried to regularly check m_Net->IsRunning() to see if I can detect this. But so far I wasn't successful.

rsta2, 2021-11-02:
Thanks @hpingel, for responding.

Yes, `CScheduler::Yield()` has to be called sufficiently often, so that the whole task system can run. I understand, that the situation in Sidekick64 is special, because it is very time critical. As I see it, the reconnect issues in your project may be caused by not calling `CScheduler::Yield()` for some time. Of course, this is not the intended operation, but I understand why you have to do it. I will think about, if there can be methods to suspend the network task for a while and resume it later.

There is a new method `CWPASupplicant::IsConnected()`, which returns if there is an active connection to an access point. `CNetSubSystem::IsRunning()` does not work for this purpose.

rsta2, 2021-11-04:
@hpingel I have added two new methods `CTask::Suspend()` and `CTask::Resume()`, which can be used to suspend a task from running for a while and to continue running it later. Furthermore there is the new method `CScheduler::GetTask(name)` with which you can request the address of the `CTask` objects for the network subsystem (task name "net") and for WPA supplicant (task name "wpa_supplicant"). These changes are at the branch *sched-improved* for the moment. Perhaps this helps to get more control over the network tasks in your project.

rsta2, 2021-11-22:
The recent changes to the scheduler have been merged to the *develop* branch now. The branch *sched-improved* has been removed.

rsta2, 2021-12-02:
The changes to the scheduler are part of the new release Circle 44.3.