rsta2/circle #94: Question about OTA update support

robdobsn, 2019-02-23:
I've been looking at moving to Circle for a retro-computer project:
https://github.com/robdobsn/PiBusRaider and https://robdobson.com/2018/09/raid-the-bus-and-the-speccy-lives/

I'd really like to get OTA updates working though (as the Pi is interfaced via an ESP32 - to get WiFi support primarily) and so the dev cycle is tedious without indirect firmware updates. I have OTA working in a hacked-together framework based mainly on DWelch and other's work but I can't seem to get it working in Circle. Probably something to do with memory management / caches but I'm unsure. Do you have any ideas for getting OTA working. I don't need the OTA to update the SD card image as I'm just thinking about development cycles. So essentially my problem is just the part where the new image (which has already been downloaded and is somewhere else high-up in memory) is swapped with the current code and executed.

rsta2, 2019-02-24:
Does this mean, you want to copy the new kernel image over the old one in memory and jump to the entry point? Unfortunately this will not work, because you would have to reset the hardware of all peripherals back to their initial state, which would be difficult. I think, the only possible solution for an update of the kernel image is writing it out to the SD card and call reboot().

rsta2, 2019-02-28:
It's working! First I have to apologize, but I guessed it is not possible without big modifications. But now, because the new Circle release has been done, I had more time to think about it and I find a quick solution for an in-memory chain boot update of the kernel image.

There is a new feature branch "chain-boot" with the required updates and a sample/38-bootloader, which shows, how the update can be done. The bootloader is using TFTP and you need a TFTP client program on your host computer to try it. For further info see the README file in the sample directory.

In case you want to include this in-memory update support into your own program, please note that it will only work if some conditions will be met:

* no multi-core support
* no accelerated graphics support
* does not work with AArch64 yet
* maybe more (which I don't know yet)

Basically you have to call EnableChainBoot() (from <circle/chainboot.h>) with a pointer to the new kernel image in memory and its size and return with EXIT_REBOOT from main(). Please note that the main.cpp file of the sample is slightly different from the normally used version.

robdobsn, 2019-03-02:
Wow that's amazing. Thanks so much for thinking about this. I think it is a really valuable feature to include, especially for rapid development, as the regular options for firmware update are quite frustrating to me. In the project I'm working on I've implemented an extension to the project's own API to allow firmware update but the approach you suggest sounds better. I'll need to explore more.

rsta2, 2019-03-04:
You are welcome. Yes, that's a really interesting feature. Chain boot is working with AArch64 too now. There has been a problem with the RPi 1, which has been fixed.

rsta2, 2019-03-08:
The chain-boot feature has been completed for the moment. I have added a web front-end to *sample/38-bootloader* too. The *chain-boot* branch was merged into *develop* and deleted.

rsta2, 2019-03-20:
This feature is now part of the new release Step39.1. So this can be closed.

closed 2019-03-20 by rsta2