rsta2/circle #95: Issue with interrupt driven serial when REALTIME is not defined

robdobsn, 2019-02-23:
I have been struggling to get a reliable serial connection working at 115200 which I didn't expect to be a problem at all given the processing power of a Pi and the lack of OS to get in the way. However I've discovered that the standard situation is for any screen access to disable interrupts while an entire string is being written and this causes the UART to drop characters. It seems odd to me to have to set REALTIME just to get a serial connection to work.

rsta2, 2019-02-24:
Yes, I know that this situation is not optimal. Doing outputs via CScreenDevice while you need an accurate IRQ timing does normally not work, especially when the screen is scrolled. That is because CScreenDevice::Write() may be called from IRQ_LEVEL too and this would crash the system, if it happens while another call to this method is already running from TASK_LEVEL. Therefore a spin lock is acquired while the screen output is running and it raises the level to IRQ_LEVEL. I didn't find a better way to implement this so far without queuing the screen output, which would make it more complex (also for the user) and I want the screen output being synchronous.

If you prefer doing screen outputs with accurate IRQ timing, you can enable the REALTIME setting. The level remains on TASK_LEVEL then, but CScreenDevice::Write() must not be used from IRQ_LEVEL then any more, which should happen only on error. Another solution would be to use the FIQ for the serial interface. The FIQ has a higher priority compared to IRQ and may be handled even while CScreenDevice::Write() is running.

robdobsn, 2019-03-02:
Thanks for the clear explanation. I think the FIQ approach may work for me - I'll try.

closed 2019-03-02 by robdobsn