rsta2/circle #97: circle/types.h vs stddef.h vs sys/types.h

thepyper, 2019-03-22:
Hi!
I have an issue compiling some code with circle, what is happening to me is that I get some double type definitions. What I'm seeing in particular is that using following toolchain:

arm-none-eabi-g++.exe (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]

I'm getting the following type redefinition errors:

...
h:\works2\union\016_907.qcpy\test\rpi_soem3\circle-stdlib\libs\circle\include\circle\types.h:49:24: error: conflicting types for 'size_t'
 typedef unsigned long  size_t;
                        ^~~~~~
In file included from h:\works2\union\016_907.qcpy\test\rpi_soem3\circle-stdlib\libs\circle-newlib\newlib\libc\include\sys\cdefs.h:45,
                 from h:\works2\union\016_907.qcpy\test\rpi_soem3\circle-stdlib\libs\circle-newlib\newlib\libc\include\stdio.h:35,
...
h:\qt\gcc-arm-none-eabi-8-2018-q4-major-win32\lib\gcc\arm-none-eabi\8.2.1\include\stddef.h:216:23: note: previous declaration of 'size_t' was here
 typedef __SIZE_TYPE__ size_t;
                       ^~~~~~
					   
...
h:\works2\union\016_907.qcpy\test\rpi_soem3\circle-stdlib\libs\circle\include\circle\types.h:50:16: error: conflicting types for 'ssize_t'
 typedef long   ssize_t;
                ^~~~~~~
In file included from h:\works2\union\016_907.qcpy\test\rpi_soem3\circle-stdlib\libs\circle-newlib\newlib\libc\include\stdio.h:61,
...
h:\works2\union\016_907.qcpy\test\rpi_soem3\circle-stdlib\libs\circle-newlib\newlib\libc\include\sys\types.h:200:18: note: previous declaration of 'ssize_t' was here
 typedef _ssize_t ssize_t;
                  ^~~~~~~

I'm not sure I should be reporting this there, just asking if I'm doing something wrong there.
I think first one is double type definition against toolchain, and second one is double type definition against newlib (in circle-stdlib).
Is there some correct way to correct this, or should I just patch my files locally?

Thanks!

thepyper, 2019-03-22:
I think I solved half of the issue.. by adding some guards (that are compatible with the ones defined in ...\newlib\libc\include\sys\types.h. In particular I added guards named _SSIZE_T_DECLARED.
The modified file is include\circle\types.h

[types.h.txt](https://github.com/rsta2/circle/files/2997379/types.h.txt)

thepyper, 2019-03-22:
Solved also second issue in a similar way...

[types.h.txt](https://github.com/rsta2/circle/files/2997447/types.h.txt)

rsta2, 2019-03-23:
This issue concerns the interaction of circle-stdlib, the toolchain libs and Circle itself. Maybe we have a problem here and we need some time for discussing this. It's good that you have found a solution for now, so that you can continue to work.

rsta2, 2019-03-24:
Can you please provide a small sample program (with Makefile), which shows these errors on build? I tried several orders of including circle/types.h, stdio.h, stddef.h and sys/types.h, but cannot reproduce the problem. Thanks!

thepyper, 2019-03-24:
Ok next days I'll find some time to make an example..

My impression is that problem arise when "circle/types.h" is included last. Because it lacks guards against previous definitions, that instead are present in both sys/types.h and stddef.h; infact I solved adding guards in circle/types.h .. to me it sounds correct in my case, I don't know how to exaustively check for that kind of issue.. looking in stddef.h about size_t it's pretty evident that a standard is not existing.. there's maaany different guards just for size_t

rsta2, 2019-03-24:
Yes, I tried to include circle/types.h last, but got no error. The guards are right for your toolchain, but are they OK for any toolchain? As you wrote, there are many different names of the guard defines. That's the reason, why I would prefer a solution without guards.

Normally with GNU C it is not a problem to have multiple typedef's of the same type, as long the definition is the same. The *size_t* type is defined as "unsigned int" in Circle for historical reason (for AArch32), but is normally defined as "unsigned long" in other header files. I would prefer to change the definition for *size_t* in Circle to "unsigned long" and of *ssize_t* to "long".

Perhaps this solves the problem without needing guards. But I can test this only, if I can reproduce the initial problem, which I cannot at the moment.

thepyper, 2019-03-25:
Hehe of course my solution is only guaranteed to work for my toolchain / system.. no standard there.
To reproduce the problem with bare minimum I did like that:

1. I use toolchain downloaded from there:
https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu-rm/8-2018q4/gcc-arm-none-eabi-8-2018-q4-major-win32.zip
I found this toolchain following link from circle's readme, and infact circle builds correctly with zero issues and circle-stdlib too.

2. I work in msys2-mingw32 shell.

3. I have built circle-stdlib from scratch, with following options:
./configure -r 3 -s /h/Qt/gcc-arm-none-eabi-8-2018-q4-major-win32/lib/gcc/arm-none-eabi/8.2.1/include
(I specify path so you see exact version of stddef.h include file)

4. From circle-stdlib root directory, I compile "ugly_type.c" with following command line:
arm-none-eabi-gcc ugly_type.c -c -o ugly_type.o -I ./libs/circle/include

5. I obtain following output:
`In file included from ugly_type.c:2:
./libs/circle/include/circle/types.h:49:24: error: conflicting types for 'size_t'
 typedef unsigned long  size_t;
                        ^~~~~~
In file included from h:\qt\gcc-arm-none-eabi-8-2018-q4-major-win32\arm-none-eabi\include\sys\cdefs.h:47,
                 from h:\qt\gcc-arm-none-eabi-8-2018-q4-major-win32\arm-none-eabi\include\stdio.h:35,
                 from ugly_type.c:1:
h:\qt\gcc-arm-none-eabi-8-2018-q4-major-win32\lib\gcc\arm-none-eabi\8.2.1\include\stddef.h:216:23: note: previous declaration of 'size_t' was here
 typedef __SIZE_TYPE__ size_t;
                       ^~~~~~
In file included from ugly_type.c:2:
./libs/circle/include/circle/types.h:50:16: error: conflicting types for 'ssize_t'
 typedef long   ssize_t;
                ^~~~~~~
In file included from h:\qt\gcc-arm-none-eabi-8-2018-q4-major-win32\arm-none-eabi\include\stdio.h:61,
                 from ugly_type.c:1:
h:\qt\gcc-arm-none-eabi-8-2018-q4-major-win32\arm-none-eabi\include\sys\types.h:200:18: note: previous declaration of 'ssize_t' was here
 typedef _ssize_t ssize_t;
                  ^~~~~~~
`

[ugly_type.c.txt](https://github.com/rsta2/circle/files/3002261/ugly_type.c.txt)

rsta2, 2019-03-25:
Thanks for your detailed description. With it I was able to reproduce these error messages, but it's not a Circle or circle-stdlib issue.

You have to specify at least the `-DAARCH=32` option, when compiling your sources, better the whole `$(ARCH)` options from *circle-stdlib/Config.mk*. Otherwise it is not possible to choose the right definition of *size_t* and *ssize_t* in *circle/types.h*.

You should also specify `-I ./install/arm-none-circle/include` as include path. [This](https://github.com/smuehlst/circle-stdlib/blob/master/samples/04-std/Makefile) can be used as example for a Makefile.

BTW I was wrong in case of the definition of *size_t* and *ssize_t* in *circle/types.h* in my last comment. It is right to define them as "unsigned int" and "int" and Circle is doing this with `-DAARCH=32`, so there is no need for a change in Circle.

thepyper, 2019-03-28:
Ok, adding this flag solved my issue!

Thanks for the info :)

rsta2, 2019-03-28:
You are welcome. It's good, that it's working now. So this issue can be closed.

closed 2019-03-28 by rsta2