rsta2/circle #101: CScreenDevice do not works with CTimer

FilipeChagasDev, 2019-07-15:
CScreenDevice.SetPixel and CScreenDevice.Write methods do not works after the CTimer initialization in kernel

rsta2, 2019-07-16:
You can test, if CTimer generally works with your QEMU, by using *sample/04-timer*. For some QEMU versions, you have to disable the system option *USE_PHYSICAL_COUNTER* in *include/circle/sysconfig.h* so that it works.

I guess, you do not have the CLogger class instantiated in your application? CTimer needs this class, because it produces logger messages. Please have a look into *sample/04-timer* for info about in which order the classes CInterrupt, CLogger and CTimer have to be constructed and initialized.

If you do not want to see the output from the logger, you still need to use the CLogger class, but you can use the class CNullDevice to create a null target for the logger.

FilipeChagasDev, 2019-07-18:
I had the problem with the "sample / 04-timer" itself. The command I'm using to run QEMU is: qemu-system-aarch64 -machine raspi3 -kernel kernel8.img.
I removed the #define USE_PHYSICAL_COUNTER from the include / circle / sysconfig.h file, but when I tested it, there was no sign of life (nothing happened). Before, there was at least a first internal log of the library (and  nothing happened after it).
Note: I changed arm-eabi-g ++ to aarch64-linux-gnu-g ++ in the Rules.mk file, since I did not find arm-eabi-g ++ with apt search.
Note2: Ignore the fork I made of the circle (Sphere), I'm not working on it.

FilipeChagasDev, 2019-07-18:
I updated QEMU and now the 04-timer application works, but only after several minutes have elapsed after the first circle log. Is it normal?

rsta2, 2019-07-18:
What do you mean with "only after several minutes have elapsed after the first circle log"? Can you explain this more.

I tried *sample/04-timer* here in QEMU 3.1.0 and everything works fine:

[image]

BTW. The recommended toolchain is noted in the AArch64 section of the main Circle README.md.

FilipeChagasDev, 2019-07-18:
"logger: Circle 39.2 started on Raspberry Pi 3 Model B (AArch64)"
This message appears immediately after the initial QEMU. The next messages, however, take several minutes to start appearing. I believe it's a QEMU problem. What command do you use to test kernel8.img?

rsta2, 2019-07-18:
I used the same command line as you mentioned above: `qemu-system-aarch64 -machine raspi3 -kernel kernel8.img`. I think, with *qemu-system-aarch64* the system option *USE_PHYSICAL_COUNTER* must be defined. Some QEMU versions need it, others don't need it. :(  Which QEMU version do you use?

FilipeChagasDev, 2019-07-18:
>> qemu-system-aarch64 --version
QEMU emulator version 3.1.50 (v3.1.0-1218-gad7a21e812-dirty)
Copyright (c) 2003-2018 Fabrice Bellard and the QEMU Project developers

rsta2, 2019-07-18:
Probably really a QEMU issue. I downloaded [QEMU version 4.0.0](https://download.qemu.org/qemu-4.0.0.tar.xz) and built it with:

```
cd qemu-4.0.0
mkdir build
cd build
../configure --target-list=aarch64-softmmu
make -j
cd aarch64-softmmu
./qemu-system-aarch64 --version
```

Then I did run it with `qemu-system-aarch64 -machine raspi3 -kernel kernel8.img` with the kernel image built from *sample/04-timer* and everything works fine. The sample waits about 5 seconds before the first logger line is displayed. On a native RPi the Act LED blinks here. Then the second logger line is displayed after 2.5 additional seconds here (depends on your PC's speed). A minute delay would be much too long.

Perhaps you can try QEMU 4.0.0?

FilipeChagasDev, 2019-07-18:
Ok, i will try to do it

rsta2, 2019-09-04:
@FilipeChagasDev Did you have an opportunity to test QEMU 4.0.0 with Circle?

FilipeChagasDev, 2019-09-08:
I couldn't even compile QEMU. My computer does not have enough RAM :(

rsta2, 2019-09-08:
That's a problem. :( But I think, you need an other QEMU for a successful test.