rsta2/circle #102: WS2812 PWM Compatibility

walkercam, 2019-07-20:
HI Rene. I am endeavoring to extend Circle to control WS2812 leds via PWM instead of SPI as in the addon example. I have been using this Linux userspace library as a reference https://github.com/jgarff/rpi_ws281x . I have known good hardware (Rpi 3b+ and a string of 12 WS2812 leds on GPIO 18) that work well with the above library. 

Now the issue is that I am just not experienced enough to successfully port the PWM/DMA component of the above library to work with Circle. I have spent days reading datasheets and others code but am still not even sure where to begin. Would you be willing to help with this? It looks like modifying the CPWMSoundBaseDevice class would allow this relatively easily.

Many thanks.

rsta2, 2019-07-21:
I will see, what I can do. But it will take some time, because I'm busy with the RPi 4 support at the moment.

walkercam, 2019-07-21:
Yes, that's understandable. I have taken the PWMSoundBaseDevice class and have started modifying it to suit this purpose. I hope I now understand enough to get a proof-of-concept working. I may need to call on your help to tidy it up and it could then be incorporated into Circle.

rsta2, 2019-07-22:
If you have specific questions, you can ask them here. Adding PWM support to *addon/WS28XX/* in Circle would be good. I can help to do that.

rsta2, 2019-08-04:
How is the status of your WS2812 PWM porting project? I would like to have a little break from my RPi 4 porting efforts now and could have a look at this.

walkercam, 2019-08-05:
I have a working prototype driving a string of WS2812 leds using one channel of the PWM peripheral. I have also been busy with other projects and have not progressed it past the prototype stage. My plan was to build a single library (essentially porting the library I linked above) to control multiple different addressable LED types (SK6812, etc) from either the SPI, PWM or PCM peripherals. Shall I share my code with you?

rsta2, 2019-08-05:
It's already working, that's good. If you want to provide your own library with that rich function, this would be great too. To integrate it into Circle requires additional work, because it should be compatible with the existing *addon/WS28XX/* library in some way. Unfortunately I cannot test it too, because I do not have the hardware. So a separate library is also an option and maybe easier to maintain. What do you think?

vanvught, 2019-10-01:
@walkercam I am just wondering; why would you not use SPI for driving the WS28xx chips?
The SPI implementation is DMA based and has no FPS limitation except for what is with the WS28xx chip protocol.

walkercam, 2019-10-03:
@vanvught that is an easy one to answer. I need SPI to communicate with other high speed peripherals in my application. I am only driving 12 WS2812 LEDs. PWM and PCM are great options that can still make use of the DMA to unload the CPU and this frees up the SPI for more important tasks.

rsta2, 2020-06-16:
There are no plans to support WS2812 LED stripes using PWM or PCM in Circle itself. So this can be closed.

closed 2020-06-16 by rsta2