rsta2/circle #104: Code stripping by default?

konistehrad, 2019-09-25:
Hello, and thank you so much for this incredible project! It's been a blast getting up to speed with using the Rpi as a microcontroller. As part of that effort, I wrote some bindings for [u8g2](https://github.com/olikraus/u8g2), but ran into a bit of a problem: the resulting binary is huge, around 10MB. This wouldn't be as much of a problem, but even at a baud rate of 460800, uploading via serial is interminably slow.

I was able, however, to enable dead code stripping quite easily during the build process! By adding `-fdata-sections -ffunction-sections` to the CFLAGS and `--gc-sections` to the LD command, the binary immediately shrank to around 60K. This is a much more reasonable size for working with the serial uploader.

All that is to say: should this be enabled by a flag? Are there side effects that these arguments bring with them that will impact other people's work? Basically, this seems like a useful feature, and I'd love to kick off a conversation on the best way to make it available to more users. Thanks again!

rsta2, 2019-09-26:
Hello and thanks! A resulting kernel image of 10MB is uncommon, but I understand that it can't be loaded via the serial interface and your solution may be a good one. Unfortunately I cannot say, if these options have side effects at the moment, but I will test it.

konistehrad, 2019-09-26:
In the case of u8g2 it makes some sense: all the font data is exported as C symbols, and [there are a lot of available fonts](https://github.com/olikraus/u8g2/wiki/fntlistall). ;) On that note: if there's a guide for how to structure and prepare an addon package for inclusion I'd love to read it before opening my PR! (I just have to test bit-banging and HW SPI integration before I'm happy with the package).

Conversely, if you're not interested in having the package part of Circle's addons please let me know and I'll spin out a new repo for it explicitly. Thanks again for this excellent project!

rsta2, 2019-09-27:
I tested the "sections flags" and it reduces the size of the kernel image as expected. Unfortunately I found samples from the [circle-stdlib project](https://github.com/smuehlst/circle-stdlib) which do not work (04-std, mbedtls/06-webclient), if build with these options. I got an exception and a failed assertion there. Other samples did work all right. I will add an option `GC_SECTIONS` to *Rules.mk*, which is not enabled by default, so one can enable it if wanted for a specific project.

The source code for a add-on library should fit into one subdirectory of addon/ to be a direct part of Circle, with a sample program in another subdirectory from there. What I have seen, the *u8g2* project is more complex, so I would prefer, if you create a new repo for it. This will also make the maintenance easier for you. Unfortunately I do not have the hardware to test this library with Circle.

rsta2, 2019-09-28:
The build option `GC_SECTIONS` has been added on the *develop* branch and will be part of the next release. Thanks!

konistehrad, 2019-09-30:
Understood on both counts! I'll keep it forked; you're not wrong, it's pretty big. Thanks for getting that in, super appreciate it!

rsta2, 2019-10-01:
OK, thanks! I think, this can be closed.

closed 2019-10-01 by rsta2

konistehrad, 2019-10-16:
For reference, I finally took the time to spruce up the demo code and drop the repo for u8g2 I2C bindings. Not sure where else to toss this link, but for tracking purposes here it is: https://github.com/konistehrad/u8g2_circle

Thanks again!