rsta2/circle #106: Use -mcpu= instead of -march= with -mtune=

vanvught, 2019-10-01:
Hi Rene,

With reference to the links below, it is interesting to work with -mcpu. It will enables hardware 32-bit idiv.

- Arjan

https://community.arm.com/developer/tools-software/tools/b/tools-software-ides-blog/posts/arm-cortex-a-processors-and-gcc-command-lines

https://stackoverflow.com/questions/42718572/gcc-mtune-vs-march-vs-mcpu

> So to answer your question - on ARM, always use -mcpu for best performance, unless you care for backwards compatibility.

rsta2, 2019-10-02:
Hi Arjan, interesting point. I think, this has most influence with `RASPPI = 2`, where it enables the hardware division instruction, as you wrote. If I use `-mcpu=cortex-a7` there, this unfortunately does not fit for the Cortex-A53 of the RPi 3. Please note, that kernel images, which have been built with `RASPPI = 2` (kernel7.img) should work on the RPi 3 too. This is the same behavior as it is used for Raspbian kernels. But perhaps kernel images built with `-mcpu=cortex-a7` do run on the Cortex-A53 too and are faster because of the hardware division. I will try it. Thanks for the references!

rsta2, 2019-10-06:
I've tested `-mcpu=cortex-a7` with `RASPPI = 2` on the Raspberry Pi 3B+ with some samples and it's working. The hardware division instruction is used, as it can be seen in the *kernel7.lst* file and kernel images are slightly smaller.

Thus I suppose, this option can be used generally and the generated *kernel7.img* images will run successfully on the Cortex-A53 too, like it is required. I have replaced the `-march= -mtune=` options with `-mcpu=` in *Rules.mk* for all targets on the *develop* branch. It will be part of the next release. Thanks for the suggestion!

closed 2019-10-09 by rsta2