rsta2/circle #108: Pi Zero W: Serial Flasher vs. Circle UART usage

konistehrad, 2019-10-16:
Hello again! I've noticed that after uploading a new kernel with the serial flasher the serial output from the kernel itself appears to be garbage data. Not sure if this is the miniUART butting heads with the PL011 or what, but I figured I'd kick off a conversation about it to see if we could get this working. Thanks!

rsta2, 2019-10-16:
Because I do not have a Zero W, I tried this with a RPi Zero: Uploading the kernel image (of sample/04-timer) with *make flash* and watching the serial output with *make monitor* afterwards. The logging output was directed to the serial device of course. It did work without problems. Currently I have no idea, why this does not work at your site.

Please note that the bootloader uses the mini-UART, while the class *CSerialDevice* in Circle uses the PL011 UART. There should be no conflict because of this, because the PL011 UART is initialized from scratch in Circle.

konistehrad, 2019-10-16:
Gotcha... so I can try this using the same setup as you, what baudrate did you have both the flasher and kernel set to?

rsta2, 2019-10-17:
I have set `FLASHBAUD = 1500000` in *Config.mk* for the bootloader. That's the maximum, which is possible with my serial USB adapter. The Circle sample programs use 115200 baud by default, so *USERBAUD* does not need to be changed.

konistehrad, 2019-10-17:
I can confirm that it works with `04-timer` but not my code, which I assume means this is my fault not yours. Closing, and thanks for your prompt response!

closed 2019-10-17 by konistehrad