rsta2/circle #110: vc4 interface samples don't display properly on Pi 4

randyrossi, 2019-12-14:
Tried both hello_dispmanx and hello_triangle on the Rpi 4 today. The kernel images run but do not display any data.  Looks like for dispmanx, the resources are still pointing to uninitialized data so you see an alternating gray pattern of 1's and 0's.  For tiger, you just see a light gray screen.

rsta2, 2019-12-15:
Unfortunately the VC4 accelerated graphics support is not available on the RPi 4. So these samples will not work. The RPi 4 has a new VideoCore6 GPU and it is currently not supported by Circle. I guess, it will be difficult to support it.

randyrossi, 2019-12-15:
I see.  Any chance for the dispmanx api? This post seems to imply it is still possible on the Pi4

https://www.raspberrypi.org/forums/viewtopic.php?t=243429

Seems most of the calls actually do work.  The layers I create show up in the right locations/dimensions, but they appear to show random data instead of what's in my buffer.   Maybe the resource copy doesn' work?  But if you think it's simply not possible I'll give up now :)

rsta2, 2019-12-16:
Yes, maybe the resource copy is wrong. I will have a look on it. Will take some time.

rsta2, 2019-12-16:
That went quicker than expected. I think, I have found the problem. There is a new 36-bit-enabled page list format for bulk transfers in VCHIQ on the RPi 4. I have updated the driver according to this and the *hello_dispmanx* sample works now. Can you please test, if this works with your own application?

randyrossi, 2019-12-16:
It works!  Flawlessly!  Thanks so much!

BTW, this is my project.  http://accentual.com/bmc64/

rsta2, 2019-12-16:
You are welcome. Thanks for testing. I already had a look at your awesome project. ;)

closed 2020-01-31 by randyrossi