rsta2/circle #111: Cancel in ViceSoundBaseDevice can cause infinite loop

randyrossi, 2020-01-31:
https://github.com/rsta2/circle/blob/b3a424998907f4b6381b7d2bd9bd5569ae5a9c5c/addon/vc4/sound/vchiqsoundbasedevice.cpp#L250

This line can cause an infinite loop if there is no data left to be dequeued.  The state will never transition to cancelled.  I replaced this section with this code below and cancel will work if there is no data enqueued at the time cancel is called.

```
  m_State = VCHIQSoundCancelled;
  if (m_nWritePos - m_nCompletePos > 0) {
     while (m_State == VCHIQSoundCancelled) {
       CScheduler::Get()->Yield();
     }
  } else {
     m_State = VCHIQSoundTerminating;
  }
```

rsta2, 2020-02-01:
This makes sense. Thanks for the patch! It will be part of the next release.

rsta2, 2020-02-04:
Circle 41 is out with this patch included

closed 2020-02-04 by rsta2