rsta2/circle #112: Unaligned gencmd_client field causing crash with vc_gencmd

randyrossi, 2020-02-09:
https://github.com/rsta2/circle/blob/d2cc5335229a7ecb889ecd6a8d93a5acde79426e/addon/vc4/interface/vmcs_host/vc_vchi_gencmd.c#L67

I was trying to make my own calls to gencmd and noticed that some calls were crashing if the response was larger.  I traced the problem down to this field not being aligned to a four byte boundary due to the GENCMD_MAX_LENGTH+1 in the preceding field.  (That length is 512 so the +1 caused response_buffer to be on an odd address).

This caused a crash in the memcpy when the response was dequeued in vchiq_shim.c here (https://github.com/rsta2/circle/blob/d2cc5335229a7ecb889ecd6a8d93a5acde79426e/addon/vc4/vchiq/vchiq_shim.c#L349)

I'm not exactly sure why the memcpy was failing because I didn't think it needed addresses to be on 4 byte boundaries.  Perhaps it's trying to optimize with larger sizes and it's not covering all cases?

When I changed GENCMD_MAX_LENGTH to 511 instead of 512, I was able to make gencmd calls that had larger response buffers so I'm pretty sure it wants some sort of alignment.

rsta2, 2020-02-10:
Unfortunately I cannot reproduce the problem on a Raspberry Pi 4B AArch32 using Circle 41. There is a test program attached to this comment, which runs successfully here. Can you please modify the program, so that the crash occurs? It builds like the *hello_dispmanx* sample.

`memcpy()` is prepared to work with unaligned buffers. Both source (r1) and destination (r0) are checked [here](https://github.com/rsta2/circle/blob/master/lib/util_fast.S#L34). If one of them is not aligned, copying takes place byte by byte.

[vc_gencmd_test.zip](https://github.com/rsta2/circle/files/4179739/vc_gencmd_test.zip)

rsta2, 2020-02-16:
I had a look into the generated code now and it turns out, that the `response_buffer` is located at an aligned address by the compiler, even with `GENCMD_MAX_LENGTH` set to 512:

```
00009ad4 <vc_gencmd_read_response>:
...
    9b50:	e59f108c 	ldr	r1, [pc, #140]	; 9be4 <vc_gencmd_read_response+0x110>
    9b54:	e1a03008 	mov	r3, r8
    9b58:	e3a02b01 	mov	r2, #1024	; 0x400
    9b5c:	e58d7000 	str	r7, [sp]
    9b60:	e2844001 	add	r4, r4, #1
    9b64:	e4950004 	ldr	r0, [r5], #4
    9b68:	eb003c82 	bl	18d78 <vchi_msg_dequeue>
...
    9be0:	0009c8c8 	.word	0x0009c8c8
    9be4:	0009c4c8 	.word	0x0009c4c8
    9be8:	0009c4cc 	.word	0x0009c4cc
    9bec:	0009c8d0 	.word	0x0009c8d0
```

The address of `response_buffer` is loaded at 0x9b50 from 0x9be4 and is 0x0009c4c8, which is an aligned address. This cannot be an un-alignment issue and the crash must have an other cause.

randyrossi, 2020-02-16:
This must be a side effect of something in my build then.  Sorry, I haven't had a chance to try your test program yet.  I will do that tomorrow.  I'll also try the test program with my sysconfig.h to see if that makes a difference.  My build does compile that to an unaligned address and I'll have to find out why.  Even then, I didn't think it should have caused problem.

randyrossi, 2020-03-28:
Finally had a chance to try the test app.  It works fine so it's something in my local changes that did this.  Closing.

closed 2020-03-28 by randyrossi