rsta2/circle #113: How to enable Falling Interrupt?

saucompeng, 2020-03-16:
I want to create a 16Mhz clock and get samples from Gpio at the falling edge of the clock.

What should the trigger parameters be?

https://github.com/rsta2/circle/blob/f4b33cc282863973e492214341565a4b7f09594f/sample/11-gpioclock/kernel.cpp#L109

rsta2, 2020-03-16:
`Sampler()` can only sample at maximum speed or with a timed delay, but not on an edge at some GPIO pin. The trigger works only for delaying the start of the sampling process, not on each sample. So it cannot be used for your purpose.

Unfortunately the maximum sample speed is about 16 MHz with only one GPIO read at a time on all Raspberry Pi models. There is no time for additional reads to check for a falling edge. It guess, the Raspberry Pi GPIO block is not quick enough for your purpose.

rsta2, 2020-03-21:
I think, this question has been answered and the issue can be closed.

closed 2020-03-21 by rsta2