rsta2/circle #114: AARCH64 and -O3 on Pi3B image does not get past rainbow boot screen

dsvf, 2020-03-26:
Hi!

In Rules.mk, I set OPTIMIZE to O3, AARCH to 64 and RASPPI to 3, then I run makeall clean && makeall. The resulting kernel8.img does not get past the initial "Rainbow" screen.
The samples work successfully when using either -O2 AARCH64 or -O3 AARCH32.

I solved this for my application by adding OPTIMIZE = -O3 to the sample makefile. Is that an acceptable way to set the optimization level (or other flags, for that matter) for a certain source file only?

rsta2, 2020-03-27:
There must be an incompatibility with -O3 in the early stages of AArch64 initialization. I will put this on my TODO list, but it will take time to sort it out. There is no problem with compiling different source files with different optimizing options, so your way to use -O is acceptable.

rsta2, 2020-04-08:
Has been fixed on the *develop* branch.

rsta2, 2020-05-09:
Should be fixed in Circle Step42. So close.

closed 2020-05-09 by rsta2