rsta2/circle #115: sample 27-usbgamepad crashes/stalls using Kiwitata gamepads

randyrossi, 2020-03-28:
Hi.  I have two Kiwitata brand gamepads (cheap).  These seem to cause circle's usb stack to report gamepad status many thousands of times per second (through RegisterStatusHandler callback) unlike my other Logitech gamepad which only reports status when buttons or axes actually change.  So when I run the sample app, it spits out status repeatedly even if nothing changes.

On the Pi0, the sample app 27-usbgamepad will crash/stall if I have both plugged through a powered USB hub.

I get this msg repeated:  **dwhci: Transaction failed (status 0x82)**

Then the sample app seems to stall for a couple minutes, then spits out more messages before stalling again.

My power adapter is an official Rasbperry Pi adapter with 2.5A rating.  This stall/crash problem doesn't appear to happen on My Pi2/Pi3B or Pi3B+ using the same game pads.  

Any ideas?  Thanks.

randyrossi, 2020-03-28:
It's worth noting that I still get the transaction failure messages even without the USB hub.  But then only one gamepad is connected so it doesn't stall.

rsta2, 2020-03-29:
I guess you are using the `USE_USB_SOF_INTR` system option? Maybe the interrupt rate is too high with this gamepad. You could try to enable [this code](https://github.com/rsta2/circle/blob/f4b33cc282863973e492214341565a4b7f09594f/lib/usb/usbendpoint.cpp#L106) even with USE_USB_SOF_INTR defined to reduce the interrupt rate. Normally with USE_USB_SOF_INTR it is not throttled.

randyrossi, 2020-03-29:
Y, I have tried with and without USE_USB_SOF_INTR but with both joysticks plugged in, the Pi0 never makes any calls to the callback.  I think it's stuck on a usb transaction.   If I unplug one of the gamepads, it starts printing transaction failed messages so it's still alive.   I can try putting debug statements to see where it gets stuck.

rsta2, 2020-03-30:
I think the only possible solution is to reduce the interrupt rate as noted in my previous comment. "Transaction failed" messages should normally not occur, because the USB driver can recover from this only in a few cases. Please note that there is no datasheet available for the DWHCI controller of the RPi so it is difficult to make it work with any USB device in any case.

rsta2, 2020-10-27:
I've added the Kiwitata gamepads to the list of incompatible devices in the project wiki.

closed 2020-10-27 by rsta2