rsta2/circle #117: Random USB failures on Raspberry Pi 4

doraemoncito, 2020-04-13:
I'm trying to build an application using Circle and I have found random failures in the USB subsystem when I compile the code for Raspberry Pi 4.  Even when the code doesn't change, the output can be quite different from one run to the next.  e.g.

Run #1:

logger: Circle 41.2 started on Raspberry Pi 4 Model B
00:00:00.66 timer: SpeedFactor is 1.51
00:00:00.73 pcie: Link up, 5.0 Gbps x1
00:00:01.09 usbdev1: Device ven2109-3431, dev9-0-1 found
00:00:01.10 usbdev1: Interface int9-0-0 found
00:00:01.10 usbdev1: Using device/interface int9-0-0
00:00:01.68 xhciroot: Port 1: Device configured
00:00:01.71 kernel: ZX Raspberry: a bare metal Sinclair ZX Spectrum emulator for Raspberry Pi
00:00:01.72 kernel: Copyright (c) 2020 Jose Hernandez
00:00:01.72 kernel: Build date: Apr 13 2020 10:43:28
00:00:01.73 kernel: Keyboard not found

Run #2:
logger: Circle 41.2 started on Raspberry Pi 4 Model B
00:00:00.66 timer: SpeedFactor is 1.51
00:00:00.73 pcie: Link up, 5.0 Gbps x1
00:00:01.10 usbdev1: Device ven2109-3431, dev9-0-1 found
00:00:01.10 usbdev1: Interface int9-0-0 found
00:00:01.10 usbdev1: Using device/interface int9-0-0
00:00:01.83 usbdev1-2: Device ven603-f2 found
00:00:01.84 usbdev1-2: Interface int3-1-1 found
00:00:01.84 usbdev1-2: Using device/interface int3-1-1
00:00:01.85 usbdev1-2: Interface int3-0-0 found
00:00:01.85 usbdev1-2: Using device/interface int3-0-0
00:00:01.91 xhciendpoint.cpp(305): stack[1] is 0x2C5C4
00:00:01.92 xhciendpoint.cpp(305): stack[15] is 0x2F598
00:00:01.92 xhciendpoint.cpp(305): stack[16] is 0x10000
00:00:01.93 xhciendpoint.cpp(305): stack[18] is 0x11020
00:00:01.93 xhciendpoint.cpp(305): stack[25] is 0x10000
00:00:01.94 xhciendpoint.cpp(305): stack[28] is 0x2FA44
00:00:01.94 xhciendpoint.cpp(305): stack[31] is 0x2F074
00:00:01.95 xhciendpoint.cpp(305): stack[39] is 0x2DDE8
00:00:01.95 xhciendpoint.cpp(305): stack[47] is 0x3ED18
00:00:01.96 xhciendpoint.cpp(305): stack[60] is 0x2FA44
00:00:01.96 xhciendpoint.cpp(305): assertion failed: (uintptr) pBuffer > MEM_KERNEL_END

If I run the USB keyboard example that comes with Circle, I find that sometimes it just works and sometimes it doesn't detect the keyboard at all.  Again, all without changes, just powering down the Pi and starting it again shows this behaviour.

Any suggestions?

rsta2, 2020-04-14:
Can you create a file *cmdline.txt* on your SD card with the option `usbpowerdelay=2000` and test again? Further info about this option is [here](https://github.com/rsta2/circle/blob/master/doc/cmdline.txt#L21). To analyze the reason for the failed assertion I would need the file *kernel7l.lst* (result of the build process) and the matching stack trace. I wonder if you could send them (compressed) to me by Email?

doraemoncito, 2020-04-14:
Hi, I've added the USB power delay parameter to the command line in file using the USB keyboard example from Circle and restarted the Pi 4 half a dozen times and it seems to work.

In my own application, that change had no effect and I am still seeing the error:

```console
logger: Circle 41.2 started on Raspberry Pi 4 Model B
00:00:00.66 timer: SpeedFactor is 1.51
00:00:00.67 kernel: Initialising USB subsystem
00:00:00.73 pcie: Link up, 5.0 Gbps x1
00:00:02.80 usbdev1: Device ven2109-3431, dev9-0-1 found
00:00:02.80 usbdev1: Interface int9-0-0 found
00:00:02.81 usbdev1: Using device/interface int9-0-0
00:00:05.03 usbdev1-4: Device ven603-f2 found
00:00:05.03 usbdev1-4: Interface int3-1-1 found
00:00:05.04 usbdev1-4: Using device/interface int3-1-1
00:00:05.04 usbdev1-4: Interface int3-0-0 found
00:00:05.05 usbdev1-4: Using device/interface int3-0-0
00:00:05.11 xhciendpoint.cpp(305): stack[1] is 0x323F4
00:00:05.11 xhciendpoint.cpp(305): stack[15] is 0x38D00
00:00:05.12 xhciendpoint.cpp(305): stack[16] is 0x10000
00:00:05.12 xhciendpoint.cpp(305): stack[18] is 0x11020
00:00:05.13 xhciendpoint.cpp(305): stack[25] is 0x10000
00:00:05.13 xhciendpoint.cpp(305): stack[28] is 0x391AC
00:00:05.14 xhciendpoint.cpp(305): stack[31] is 0x387DC
00:00:05.14 xhciendpoint.cpp(305): stack[39] is 0x37550
00:00:05.15 xhciendpoint.cpp(305): stack[47] is 0x48480
00:00:05.15 xhciendpoint.cpp(305): stack[60] is 0x391AC
00:00:05.16 xhciendpoint.cpp(305): assertion failed: (uintptr) pBuffer > MEM_KERNEL_END
```

I'll send you a compressed kernel via email as requested.

Generally, I deploy my kernel to the Pi 4 using U-Boot and TFTP so using the command line approach won't work in that setup.  Is there a way of replicating the effect of the USB power delay parameter programmatically in the kernel so that it can be deployed with U-Boot?

rsta2, 2020-04-14:
Thanks for sending the kernel image file (kernel7l.img). But unfortunately I need the kernel7l.lst file (the assembly listing), not the kernel image itself. I need the assembly listing to see, which functions where called  before the assertion fails.

I guess, there are two different problems: One is the failed detection of your keyboard, which seems to be solved by the "usbpowerdelay" option. The other is the assertion, which is probably caused by using a buffer inside static data segments (.data or .bss) of the kernel for USB transfers, which is not allowed with the Circle xHCI driver for the RPi 4. All buffers used for USB transfers must be allocated on the heap or on the stack.

Perhaps you have an idea, what happens in your application with this explanation. If not, please send the kernel7l.lst (!) file so I can look for it.

Please note, that the "usbpowerdelay" option should work, even when you use U-Boot, if the cmdline.txt file is on the SD card.

doraemoncito, 2020-04-14:
Heres the assembly listing:

doraemoncito, 2020-04-16:
Thanks Rene for your help pinning down the issue.  In summary, the solution to get my application running is twofold:

- Create a file cmdline.txt at the root the SD card with the option usbpowerdelay=2000.  This option works even when you use U-Boot to download the kernel from a TFTP server. Further info about this option can be found [here](https://github.com/rsta2/circle/blob/master/doc/cmdline.txt#L21). 

- Increase the stack size by a multiple of 16K and rebuild circle and the target application / kernel.  In my particular case I changed the [stack size](https://github.com/rsta2/circle/blob/master/include/circle/memorymap.h#L42) to 0x100000. I.e.

```
diff --git a/include/circle/memorymap.h b/include/circle/memorymap.h
index e96f920..729a388 100644
--- a/include/circle/memorymap.h
+++ b/include/circle/memorymap.h
@@ -39,7 +39,8 @@
 
 #define PAGE_SIZE              4096                            // page size used by us
 
-#define KERNEL_STACK_SIZE      0x20000                         // all sizes must be a multiple of 16K
+#define KERNEL_STACK_SIZE      0x100000                        // 64 * 16K
+//#define KERNEL_STACK_SIZE    0x20000                         // all sizes must be a multiple of 16K
 #define EXCEPTION_STACK_SIZE   0x8000
 #define PAGE_TABLE1_SIZE       0x4000
 #define PAGE_RESERVE           (4 * MEGABYTE)
```

closed 2020-04-16 by doraemoncito