rsta2/circle #118: Synchronous Exception on multi-core TCP/IP application?

rbellek, 2020-04-29:
Hi,

I want to create a TCP/IP application using circle with mutli-core support on Raspberry Pi 4 Model B.

Basicly, I want to read GPIO pins write the data into a ring buffer, and send the data through TCP/IP socket.

Core 0 is dedicated for GPIO read/write ops. (gpio task)
Core 1 is dedicated for TCP/IP socket. (tcp task)
Core 2 and 3 are shutdown by SendIPI(nCore, IPI_HALT_CORE) command.

I use CSpinLock for sync. For simplicity, the beginning of both task m_lock.Acquire() is called and of course end of the both task m_lock.Release() is also called.

The problem is when the data start to fill the ring buffer and the tcp task is try to send data a system exception is occurs:

rsta2, 2020-04-30:
Please modify the core assignment as follows:

* Core 0 runs the TCP/IP stack (task)
* Core 1 runs GPIO read/write (no task)
* Core 2: return from CMultiCoreSupport::Run()
* Core 3: return from CMultiCoreSupport::Run()

In Circle all tasks have to run on core 0. Because the TCP/IP stack is task based, it has to run on core 0. All other cores can run any additional function, but not a task (derived from CTask).

For core 2 and 3 you should simply return from the Run() method in the class derived from CMultiCoreSupport. They will sleep then automatically until system power-off.

Further info on multi-core support is available in *doc/multicore.txt*.

rbellek, 2020-05-01:
Yes, It works great :)

Thank you very much for your help.

rbellek, 2020-07-20:
### Kernel panic message "heaplow: Out of memory"

[image]

- No use of recursive functions (1 main while loops for each task/function)
- Only one ring buffer for data storage (HEAP_HIGH 1 GB)
- No dynamic allocation (new/malloc) expect ring buffer
- TCP socket connects only once and sends data in an infitine loop (is this causes the error?)

[image]

I got still same error message. I am not sure what is wrong. If you request I can give extra details about source code / configs.

Thank you.

rsta2, 2020-07-21:
I guess, your application is generating data more quickly than it can really be sent out to the TCP peer. Data waiting for TCP sent has to be queued in the low heap and on the long run, if more data is generated than can be sent, the low heap will overrun.

Please note that the TCP implementation in Circle is not very optimized for speed. You will not get a transfer rate of more than some MBytes/sec even on Gigabit Ethernet.

You can set the 3rd parameter of `CSocket::Send()` to 0 (instead of MSG_DONTWAIT) to ensure, that the data is sent out before `CSocket::Send()` returns.

rbellek, 2020-07-21:
[image]

I already use 0 for 3rd parameter of CSocket::Send(), still same error. In ring buffer implementation, is ring buffer is full, data writing is wait for a free slot. I also add a spinlock for concurrent accesses.

To be honest, this implementation was just worked. I got (peak) 280 mbps data bandwidth over gigabit ehternet and I have tested it over 30 minutes. I have verified (byte-by-byte) all the data received from server (pc side). I have changed some codes which (I think) is not related to write/send logic. Now application starts, send some data (approx 50 mb ~ 150 mb) and stops working.

rsta2, 2020-07-21:
Sometimes it is needed to explicitly call `CScheduler::Get ()->Yield ()` from the main loop on core 0, so that the scheduler can run the background tasks. You have to `#include <circle/sched/scheduler.h>` for this.

If this doesn't help, I can only offer to have a direct look at your source code. You can send it to my Email address on my GitHub main user page, if you want to.

rsta2, 2020-08-04:
I suppose, the problem has been fixed with a commit on the *develop* branch. It will be on *master* with the next release. The problem was caused by an invalid flow control, if the flags parameter has been 0, when calling `CSocket::Send()`. The calling task did not block long enough. With this change the TCP transfer speed may slow down a bit. One can get the full speed by using the flags parameter MSG_DONTWAIT as before.

@rbellek Thanks for sending your source code and TCP server test application and for support, while analyzing this issue! This helped very much.

rbellek, 2020-08-04:
@rsta2 Thank you very much for bug fixing. I will test the code ***in production***. I think this issue is solved. Maybe, I can send you a report about performance (if it makes a dramatic difference).

closed 2020-08-04 by rbellek