rsta2/circle #119: it canot run on RPI4

figozhang, 2020-06-03:
i have build this image for RPI4, but it cannot work, on output in PL Uart.

rsta2, 2020-06-03:
By default there is no output to the UART. Most sample programs use a HDMI display for output. You can create a file *cmdline.txt* with this contents on the SD card to direct the output to the UART:

```
logdev=ttyS1
```

On the RPi 4 there is an additional requirement, if there is no HDMI display attached and a sample program should be used with the UART. You have to define the system option `SCREEN_HEADLESS` in this case [here](https://github.com/rsta2/circle/blob/master/include/circle/sysconfig.h#L258).

figozhang, 2020-06-03:
Starting start4.elf @ 0xfec00200

logger: Circle 42 started on Raspberry Pi 4 Model B (AArch64)
00:00:00.66 timer: SpeedFactor is 1.51
00:00:00.73 pcie: Link up, 5.0 Gbps x1
00:00:01.09 usbdev1: Device ven2109-3431, dev9-0-1 found
00:00:01.10 usbdev1: Interface int9-0-0 found
00:00:01.10 usbdev1: Using device/interface int9-0-0
00:00:01.68 xhciroot: Port 1: Device configured
00:00:01.68 kernel: Compile time: Jun  4 2020 08:14:16
00:00:01.69 kernel: Keyboard not found

there is shell command for circle?

rsta2, 2020-06-04:
Circle is intended to build bare metal user applications and comes with a number of sample programs, which often show a single feature and then terminate. Most samples do not have a shell. 

But if you want to try a shell via UART, you can use [sample/32-i2cshell](https://github.com/rsta2/circle/tree/master/sample/32-i2cshell). This program provides a number of commands to control the I2C interface.

There is also a more extensive test program [unitest](https://github.com/rsta2/unitest), which is based on Circle, has a shell and shows much more features. It should work automatically via the UART, if there is no HDMI display or USB keyboard attached.

rsta2, 2020-06-16:
The initial problem has been solved and all questions have been answered. So close.

closed 2020-06-16 by rsta2