rsta2/circle #122: Relation between CSocket & Screen Message?

rbellek, 2020-06-12:
Hi all,

it might be a bit weird but is there any relation between CSocket TCP send data function and screen message print function?

I build an application using circle with multi-core support using Raspi 4 B.

- [ ] Core 0 for TCP function (!)
- [x] Core 1 for reading gpio pin & buffering data
- [x] Core 2 just return from manager
- [x] Core 3 just return from manager

I use CSocket for tcp send data.

```cpp
...
CString Message = "Debug test message";
m_pScreen->Write(Message, Message.GetLength()); // This causes Socket sent data!!!
...
if (m_manager->m_pSocket->Send(data, sizeofdata, 0) < 0)
{
....
```

If I do not write any debug message on screen csocket is not sending data to client. I try lots of time, with the same source code without changing anything but screen write line, tcp socket start to send data. ***I do not want to print messages on screen beacuse it slows down the speed of run.***

I would be very happy if you guys help me. Thank you.

rsta2, 2020-06-13:
There is no direct dependency of *CSocket* and *CScreenDevice*, this can only be a side effect. Can you please replace the debug message output with this:

```
#include <circle/sched/scheduler.h>

...

CScheduler::Get ()->Yield ();
```

This will allow other tasks to run on core 0. An explicit call of *Yield()* is needed in some cases.

rbellek, 2020-06-21:
> There is no direct dependency of _CSocket_ and _CScreenDevice_, this can only be a ***side effect***. 

I agree with that.  ```CScheduler::Get ()->Yield ();``` saved my day :) 
Thank you again.

closed 2020-06-21 by rbellek