rsta2/circle #123: Pi 4 I2C #5 register address

mjrgh, 2020-06-24:
It looks like the register base address for I2C 5 is wrong in i2cmaster.cpp (77).  It's defined there as ARM_IO_BASE + 0x205A00, but the [BCM2711 peripheral reference](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0.pdf) has it at 0x205A**80**.  I tested it with the updated address and it seems to work properly.

rsta2, 2020-06-24:
The base address of I2C5 was taken from the Linux [device tree source](https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/bcm2838.dtsi#L173), because this reference manual was not available, when the support was implemented. There is obviously a difference between these references. I will check this. Thanks for reporting this issue.

rsta2, 2020-06-24:
I checked this on a RPi 4B with 2GB RAM. It seems that the address of the extended I2C devices is not fully decoded in the SoC. Both base addresses (ARM_IO_BASE + 0x205A00 and ARM_IO_BASE + 0x205A80) belong to the same I2C5 device. When modifying the writable bits at offset 0x205A00 and reading it back, the same result is reported for the registers at the offsets 0x205A00, 0x205A80, 0x205B00 and 0x205B80. Each I2C5 register is accessible at 4 addresses.

As a conclusion there seems to be no difference, if ARM_IO_BASE + 0x205A00 or ARM_IO_BASE + 0x205A80 is used as base address for I2C5. Both should work.

mjrgh, 2020-07-06:
Just FYI, I confirmed with some attached I2C devices that it works with either address, so you're right that it doesn't really make any practical difference with the current chip set.  (If it were me, I'd probably still make the update to match the reference material, just in case a future rev of the chip set starts treating the bits as significant.)

closed 2020-07-06 by mjrgh

rsta2, 2020-07-07:
OK, I will change the base address to meet the reference in the next release. Thanks for testing.