rsta2/circle #124: DMA4 support

mjrgh, 2020-07-06:
Support for the DMA4 channels in the Pi 4 would be a nice-to-have.  I did some testing with it for memory-to-memory transfers, and it runs almost twice as fast as the special "AXI burst" mode on legacy channel 0.  It would probably be quite useful for people doing video or other performance-critical tasks.  Plus, with the library currently limited to channels 0-7, we have fewer channels available overall than on earlier Pi models.

(I'd submit a PR if I had a more complete implementation, but I only did as much as I needed for non-interrupt memory-to-memory.  As I'm sure you already know it's a bit of work to implement thoroughly, because they changed everything around from the legacy channels - different control block, different address translation rules, and worst of all, shared IRQs on channels 7&8 and 9&10.)

rsta2, 2020-07-07:
I have put this on my TODO list. Will take time.

mjrgh, 2020-07-07:
Great, thanks.

A bit of technical trivia that might be worth mentioning in the eventual comments is that a burst size of 9 seems to be fastest for memory-to-memory transfers on the DMA4 channels.  The BCM2711 peripheral reference specifically mentions that as the magic number for channel 0, but it doesn't say anything about the DMA4 channels.  I did some simple timing tests with my DMA4 code and 9 seems to be the magic number there as well.  (At least for my particular setup; I'm copying from ordinary RAM into the video frame buffer, so there might be something special about that.)

rsta2, 2020-07-08:
OK, I will consider this. Thanks for info.

rsta2, 2020-08-11:
A first version of DMA4 support using the class *CDMA4Channel* has been added to the *develop* branch. It is currently not included in the normal DMA channel allocation scheme of Circle. One has to specify the DMA channel number (12, 13 or 14; 11 is normally used by the firmware) directly to the constructor of the class. The burst size can be specified for the memory copy operations as wanted.

rsta2, 2020-08-14:
DMA4 channels can be allocated now using the normal DMA channel allocation scheme of Circle. The class *CDMA4Channel* should not be used directly by an application. It should use the class *CDMAChannel* instead and specify the ID `DMA_CHANNEL_EXTENDED` for *nChannel* in the constructor.