rsta2/circle #125: Circle 42 Networking Example with LAN + WLAN + mbedtls

hpingel, 2020-07-23:
Hi Rene,

This is not a bug report. I just wanted to let you know that I spent some time on implementing network features [into a temporary fork the code of the Sidekick64 project](https://github.com/hpingel/Sidekick64/tree/net-experiments) for Pi 3A+/3B+. Since a few days I moved on to circle-stdlib and started using mbedtls. All of this is only possible because of your good work and the excellent examples that come with circle. Also, Stephan Mühlstrasser has in the same way added nice examples to circle-stdlib so that the transition to HTTPS was quite easy. Everything is still experimental in my code. Cable based ethernet works perfectly, WLAN works pretty good (although I have to do some keep-alive requests in the Sidekick64 setting to prevent a disconnect/zombie state where a reconnect takes a couple of seconds).

Because I have to build circle-stdlib twice, once with REALTIME (for cable based LAN) and once with SDHOST for WLAN, I started hacking together some Docker images that automate the build and contain pre-built versions of circle-stdlib so that the build process is shorter if I just want to compile the application.

I had a lot of fun in doing this up to now and more fun lies ahead. Thank you for your framework!

What surprises me is that there are quite a few projects out there using Circle but there seems to be no Wiki page that summarizes the projects. Is there anything like that and I haven't found it yet?

Cheers,
Henning

rsta2, 2020-07-23:
Thank you, Henning. Well done! I do not know a wiki page, that covers Circle related projects, but I could open the project wiki on GitHub for that purpose. What do you think?

hpingel, 2020-07-30:
Yes, a Wiki page sounds good! But - let me gather some projects and a page structure first in this issue here - if you are ok with that. There is nothing worse than an empty Wiki page.

I got sidetracked by my discovery that the Open Source Fantasy retro gaming computer TIC-80 can be built with circle-stdlib for Bare Metal Raspberry. So I spent some time the last couple of days to improve my Docker based build system to compile the latest development state of TIC-80 for Raspberry Pi 2,3 and 4 (at the moment only aarch32 but it might just work with aarch64 too).

rsta2, 2020-07-31:
@hpingel That's OK. Let me know, when you are ready to initially fill the Wiki, so that I can open it.

Yes, the circle-stdlib project makes it much more easy to port existing software. I think, WLAN support should become part of the circle-stdlib project itself. I think, Stephan is currently working on a different project, but this can probably be realized later.

smuehlst, 2020-08-01:
>  I think, WLAN support should become part of the circle-stdlib project itself. 

What kind of WLAN support do you envision that could be implemented in circle-stdlib? Sockets?

rsta2, 2020-08-01:
@smuehlst Basically the constructor of the class *CStdlibAppNetwork* should get an additional parameter, which selects if WLAN or normal Ethernet is used. The initialization is a bit different for WLAN then.

Furthermore the mbed TLS samples must be adapted to set this parameter. Finally the new WLAN libraries in Circle addon must be built from the circle-stdlib Makefile and they must be added to *samples/mbedtls/Rules.mk*.

I already tried this and it worked, but currently I get a number of warnings from the *hostap* project, which hasn't been sorted out so far.

smuehlst, 2020-08-01:
@rsta2 I see. I get back to you by email when I have time for this, I will probably have some questions.

dwhinham, 2020-08-04:
Hi! :smile:

Please feel free to add my project, `mt32-pi`, to any lists you are putting together: https://github.com/dwhinham/mt32-pi/

This is a baremetal port of the MUNT Roland MT-32 synth emulator using Circle and circle-stdlib. I've really enjoyed working with Circle so far, and the project has picked up a good number of followers from retro gaming communities!

Thankyou for all your hard work on Circle, it's an exceptionally well-designed framework.
Cheers!

rsta2, 2020-08-04:
@dwhinham Thank you for offering your great project for adding to our project list and for appreciating Circle!

hpingel, 2020-08-04:
Great to see what interesting projects exist! As this ticket is related to LAN + WLAN + mbedtls, I will open up another ticket for collecting Circle based projects with the goal to ultimately fill a wiki page. ;-)

hpingel, 2020-08-05:
> @smuehlst Basically the constructor of the class _CStdlibAppNetwork_ should get an additional parameter, which selects if WLAN or normal Ethernet is used. The initialization is a bit different for WLAN then.

If I understand this correctly it would then be possible for a bare metal application to configure the type of network (LAN vs. WLAN) via a configuration file and reboot the Pi with the new setting.

For the project Sidekick64 some of the GPIO bast time-critical things only work with REALTIME setting. This means that for this particular project it would sadly not be possible to easily switch because with Circle42 there is either REALTIME or SDHOST. At the moment my knowledge about Circle is not big enough to help to resolve this situation, I can only describe the different behaviour within the application that I notice.

My planned approach currently is to have two kernel images to switch between. One kernel image is compiled with REALTIME (and also offers LAN based network if run on a Raspberry Pi 3B+). A second kernel image is compiled with SDHOST and offers a reduced feature set but WLAN for users of Pi 3A+ or users who don't want to plug a network cable into a 3B+). There needs to be a UI for the user to switch between those two kernels.

Of course, Sidekick64 is not the only project existing. I'm just pointing that out because this ticket was created with this particular project in mind.

rsta2, 2020-08-05:
> If I understand this correctly it would then be possible for a bare metal application to configure the type of network (LAN vs. WLAN) via a configuration file and reboot the Pi with the new setting without the need for re-compiling anything.

Yes, this should be possible.

For the `REALTIME` vs `USE_SDHOST` problem I have unfortunately currently no solution.

rsta2, 2020-08-11:
@hpingel There is a test version of the WLAN support at the *develop* branch of the [circle-stdlib project](https://github.com/smuehlst/circle-stdlib/tree/develop) of @smuehlst. Please have a look into the *kernel.cpp* file of a mbed TLS sample, if you are interested in. WLAN operation can be activated with a simple config change there. By default the samples still work over Ethernet.

hpingel, 2020-08-11:
@rsta2: Thank you for pointing that out! I will put that on my list of things to test. At the moment, my fork of Sidekick64 ist not yet using the class CStdlibApp as a base.

Over the last couple of days I have monitored the CPU temperature a lot when doing network requests. There was a temperature difference between different kernel images - while the Sidekick64 application "idles" at around 51°C the CPU temperature is far higher once the cable based network is active. I looked into this to find the root of the temperature increase and at the moment I concluded that once I initialize the CUSBHCIDevice and the CNetSubSystem the CPU temperature will increase by nearly 10 degrees (in 30 minutes).

My measurement (I know it's a very hot summer day....) is:
Sidekick64 "idle" state: around 51°C.
Sidekick64 + CUSBHCIDevice + CNetSubSystem + no ethernet cable connected: Temperature increase up to 59°C.
Sidekick64 + CUSBHCIDevice + CNetSubSystem + DHCP + ethernet cable + active network: Temperature increase up to 62°C. ( I have moved the CPU throttling limit up to 63°C).

My question is if this temperature increase could be less steep? Are there any power management features that the Raspbian kernel uses?

rsta2, 2020-08-11:
The Raspberry Pi OS (new name ;)) kernel automatically throttles the CPU clock, when the system load is small. I guess, this is not wanted with Sidekick64. What is theoretically possible in Circle is to reduce the polling rate for the Ethernet device, when the link is down. But currently I'm not sure, if this would decrease the temperature. 62°C sounds reasonable with a CPU running at 1400 MHz, which I guess is the case with Sidekick64.

rsta2, 2020-09-06:
WLAN support is available in the latest release of the circle-stdlib project of @smuehlst. So I think, this issue can be closed, given that it covers different topics too, which is not easy to survey.

closed 2020-09-06 by rsta2