rsta2/circle #126: ARM GNU-A toolchain not needed on Linux

forderud, 2020-08-04:
Thanks for sharing this great project! It's really amazing to be able to build & run bare metal C++ programs on my Raspberry Pi.

I just discovered while trying to build the sample projects that the ARM GNU-A toolchain download does not seem to be necessary when running on Ubuntu 20.4. What I did instead was to install:
```
apt install gcc-aarch64-linux-gnu
apt install g++-aarch64-linux-gnu
```

I then modified Config.mk to contain `PREFIX64 = aarch64-linux-gnu-` instead of `aarch64-none-elf-`. This worked like a charm, with the generated binaries booting fine on my Raspberry Pi 3 B (at least for the "17-fractal" sample I tested).

Maybe it could be worth while to update README.md with a note that the in-built toolchains can also be used?

rsta2, 2020-08-04:
Thanks for this info and for appreciating Circle! I will include this hint into the README.md for the next release as suggested. Please note that I can only test Circle using one toolchain to reduce the effort and I will remain to use the ARM GNU-A toolchain for that purpose. I will mention this in the README.

hpingel, 2020-08-04:
Be aware the curious persons will then ask if they can also build aarch32 with gcc-arm-linux-gnueabi. ;-) I might conduct some tests with my Docker based circle-stdlib build automation tools to see how the artefacts differ between none-elf and linux-gnu.

forderud, 2020-08-04:
I've now tested the same procedure from Fedora 32, and hereby confirm that the circle samples also build and run out-of-the-box on my Raspberry Pi 3 B with the in-built Fedora toolchains, after doing the following:
```
dnf install gcc-c++-aarch64
```
... as well as  modifying Config.mk to contain `PREFIX64 = aarch64-linux-gnu-`.

forderud, 2020-08-05:
I've just done some adjustments to the packages needed for Ubuntu and Fedora in the comments above.

hpingel, 2020-08-05:
If I am not mistaken, the source package for Debian/Ubuntu is called gcc-defaults. From that source package the various binary packages for gcc and g++ originate. So the version of gcc-defaults is of course a moving target between different Ubuntu versions which comes as no surprise. For example, with Debian/Ubuntu we have different releases that are still in maintenance mode.

For example, with Ubuntu 18.04 LTS (still in maintanance) you get gcc 7.4.0.
With Ubuntu 20.04 LTS you get gcc 9.3.0.
With Ubuntu 19.10 you get gcc 9.2.1.
Ubuntu 20.10 will probably have gcc 10.1.0.

https://packages.ubuntu.com/source/focal/gcc-defaults
https://packages.ubuntu.com/focal/gcc-aarch64-linux-gnu (toggle between releases on that page to see the versions)

I just wanted to point that out. It doesn't mean that it won't work.

rsta2, 2020-08-05:
As a compromise I could add the following to the *AArch64* section of README.md:

> There may be toolchain packets on certain Linux platforms (e.g. `g++-aarch64-linux-gnu` on Ubuntu or `gcc-c++-aarch64-linux-gnu` on Fedora), which are install-able using the normal platform-specific mechanism. These toolchains may work with Circle and can be a quick way to use it. You have to update the `PREFIX64=` configuration option to do so. On the other hand there is some risk that such toolchain has an ABI (Application Binary Interface), which is incompatible with Circle. In this case a built Circle sample is likely to fail running without notice. Also more complex Circle applications require support from external libraries, which must be provided by the toolchain. The only toolchain which is always tested to ensure this, is referenced in the second paragraph of this section.

forderud, 2020-08-05:
> As a compromise I could add the following to the _AArch64_ section of README.md: ...

That would be great! It's so much simpler to get started with circle if using the distro-provided toolchains 👍 

> ...On the other hand there is some risk that such toolchain has an ABI (Application Binary Interface), which is incompatible with Circle. In this case a built Circle sample is likely to fail running without notice.

Do you have something concrete in mind here? Does e.g. circle pull in any precompiled libraries with run-time dependencies to glibc or other toolchain-distributed library that could potentially be incompatible?

> Also more complex Circle applications require support from external libraries, which must be provided by the toolchain. The only toolchain which is always tested to ensure this, is referenced in the second paragraph of this section.

Maybe rephrase to instead recommend building all SW with the same toolchain to limit risk of ABI incompatibilites.

rsta2, 2020-08-05:
@forderud Maybe it's simpler to start with a distro-provided toolchain, but I think, it's also not that difficult to download the ARM GNU-A toolchain, extract it and put it into the PATH variable. It needs no further installation.

Yes, I had issues with toolchains not working with Circle. Because this may be difficult to catch, I went over to the recommended toolchain. Some Circle features need external libraries (e.g. the math lib for accelerated graphics) or header files (e.g. stdint.h for WLAN). This must be provided by the toolchain.

Please note that it is often an immense effort to test all necessary samples, when a Circle release is prepared. I can only do this with one (or better two, for AArch32 and AArch64) toolchains. To use an other toolchain is OK as long everything is working, but the users must know, that this has not been tested by myself. Tested is only the toolchain, which I have recommended.

forderud, 2020-08-05:
Thanks for the explanation @rsta2 . I'm really fine with any outcome here, as long as it's somehow documented that one can also use distro-provided toolchains.

rsta2, 2020-08-05:
@forderud OK, I will point that out.

rsta2, 2020-08-08:
A note on distro-provided toolchains is in the *README.md* on the *develop* branch and will be on *master* after the next release.