rsta2/circle #127: Creating a list of Circle based bare metal projects for the Raspberry Pi

hpingel, 2020-08-04:
With the goal to create a Wiki page at some point I'm starting to create a list of Circle based bare metal projects for the Raspberry Pi.

[work in progress]

rsta2, 2020-08-04:
@hpingel Thank you for preparing a first draft version of a list of Circle related projects! This looks already good.

What I am thinking about is, how this list can be made most useful for Circle users and how to reduce the effort to maintain it. Because maintenance will probably be required, if this list is managed in an open Wiki. For example new entries and containing links must be checked.

Given this I think (as always) focus is essential. We shouldn't try to cover all the bare metal projects for the Raspberry Pi out there, because there are so many and there are already lists on the Internet, which have a wider scope of projects. We probably won't be able to do this better than their authors and personally I don't want to spent too much time for this list. Because the Wiki is in my own project space, I think I am responsible for the contents in a way.

That means the focus should be on "Bare metal projects based on Circle or circle-stdlib", like you have listed them in the first list. I would keep other projects out of this list, including USPi related projects. For the USPi project there could be another Wiki in its own project space, if there is really demand for it. I would agree to include a very few number of links in "External resources listing bare metal projects for Pi" at the bottom of the page, because this can give a wider overview for potential readers.

hpingel, 2020-08-05:
I am absolutely ok with reducing the list to Circle based projects above to those that are most useful for learning and that are also actively maintained.

When I created the list I wanted to list everything I could find because I am very curious about in what context Circle is being used or in what context developers decide to go bare metal. One could say "retro guys like to go bare metal" but that would be a bit simple because the emulation "types" are different: I find it interesting to see - how many music related projects exist (synthesizer, Midi), how many "screen"-focussed projects exist (HDMI UI with HDMI sound). I remember there was a project doing DMX light orchestration via Circle but when I looked at it yesterday I couldn't put my finger on the software component where Circle is embededded. So I left that one out. But we can of course make a list useful to others and not only useful for my curiosity.

Regarding uspi and the "history" of developing bare metal environments: Have you documented somewhere your own personal motivation and the steps that you did towards Circle? I understood uspi as being a predecessor to Circle but it might be a spin-off instead? If uspi should be obsolete / depricated, is this documented and are the reasons already documented?

Personally I really like the Pi1541 uspi-based project and one of my daydreams is to help to convert it from uspi to circle-stdlib one day. That's why I wanted to list it. If uspi projects are not listed my motivation to port it to Circle may grow. ;-)

In short: Your suggestions are all well with me.

rsta2, 2020-08-05:
@hpingel Thanks for your understanding. Of course I understand your motivation and curiosity. I think, this list should be dedicated to help users to find useful libraries (like *circle-stdlib*) for building bare-metal applications, built on Circle or to find applications, which are using Circle, for getting suggestions, if or how some problem can be solved in your own application. It's open source and I like the idea to learn from each other. Furthermore it can be inspiring to see, which applications exist and what is possible with Circle.

My problem when creating Circle was, that I had an USB driver for the Raspberry Pi from another project, but it was needing some external infrastructure to run and I wanted to make this driver available, because the were a lot of discussion about USB back in 2014 in the RPi bare-metal forum. So I released the driver together with the needed environment and called it "Circle". This name was basically chosen, because "Pi" is the circle-number (3,14...).

The USPi project came later. There was some demand for an USB-only driver for the RPi, written in C. USPi was derived from Circle. Personally I prefer using C++ and with the growing variety of RPi hardware and growing of Circle itself, it became much more difficult to handle both projects together. I think, there is demand for a RPi 4 USB driver in USPi, but currently my motivation is not strong enough to realize this, given that it would be difficult to map some C++ features (e.g. class inheritance) to a C environment. I'm afraid the USPi driver will lose importance over the time.

rsta2, 2020-08-11:
@hpingel Do you think, the Wiki should already be created with the Circle-based projects that are listed in your comment above? Otherwise please let me know, when you are ready, so I can open the Wiki then!

hpingel, 2020-08-11:
@rsta2 : I would like to ask you to still wait a bit - I would like collect some more content for the description of the individual projects here and discuss it with you. Of course, the owners of the different projects are welcome to describe their own project.

I'm currently also involved in trying to make TIC-80 build and work again and so I'm a bit "under water" with Circle stuff:
https://github.com/nesbox/TIC-80/issues/1143

rsta2, 2020-08-11:
@hpingel No problem, there's no hurry.

hpingel, 2020-08-20:
So how to describe the Circle-based projects in a way that is helpful for developers who want to create something based on Circle?

- Describe special libraries used (for example to communicate with other hardware extensions like displays) to show examples of how to add libraries.
- Try to categorize projects.

Possible categories:
- Existing OSS software applications that are emulating or simulating a computer system, ported to a bare metal build environment. Focus is on offering an audio-visual interface using HDMI or analog Video/Audio output (human-machine-interface). Video is essential, audio is optional. Interaction via the classical/standard screen and computer keyboard. In this category we would find BMC64 and TIC-80. (Probably CPMemu? - I haven't tested it yet)
- Projects that were created based on Circle to do timing-critical GPIO-based bit banging to emulate an ancient hardware protocol (machine-machine interface). Video output via HDMI is optional. Examples: Sidekick64 (tries to obey the rules of the protocol of the expansion port of Commodore computers), [I am not supposed to name Pi1541 here as it is not Circle based ;-)].
- Projects that are focussed on emulation of audio synthesizers focussed on audio output (human-machine-interface) and emulating music keyboard features (via MIDI, USB2Midi, PC Keyboard)
- ...

I'm stopping here to let you comment.

rsta2, 2020-08-20:
I think, it's good to distinguish between libraries and applications and to categorize the applications. The categories, you have listed, are the three I also know to be the major areas of applications, based on Circle. So I agree.

farvardin, 2020-08-24:
There is this project too, which is a recreation of the zx spectrum in bare metal: http://zxmini.speccy.org/en/index.html
So far, it looks like it's closed source. The legal "THIS SOFTWARE MUST NOT BE SOLD, NEITHER ALONE NOR AS PART OF A BUNDLE.", looks like the author fears his work might be "stolen". Maybe it would be interesting to contact him and explain him the benefit of open sourcing this project (to make it more sustainable for example).

rsta2, 2020-08-24:
@farvardin I know this project and its author very well. We were working together and he helped to improve Circle very much and also contributed source code (e.g. for the gamepad support). I'm very unhappy, that the source code of this great project is not available and we already had a conversation about this over a year ago.

Yes, he fears that his work might by used by third parties to make money and in fact this already did happen. He pointed me to an offer on eBay later, where somebody sold SD cards of his ZX Spectrum emulator. I guess, this sentence was included later to prevent these offers.

I wasn't successful in explaining, why open source is good, in this conversation and I do not have contact to him any more.

rsta2, 2020-10-04:
Because it took a little bit long now, I have opened the wiki and quickly created a very basic project list in it. I think, project authors should decide by themselves, if they want their project added to this list. That's why I only added projects from authors, who already have told their interest. I created a table and have made it deliberately simple. Perhaps project classes can be added later, when more projects are in the list.

rsta2, 2020-10-27:
OK, the list is out. It's not "very successful" at the moment, but we will see. ;) I think, this issue can be closed.

closed 2020-10-27 by rsta2