rsta2/circle #130: Using circle as a submodule with REALTIME defined

robdobsn, 2020-09-01:
I'm trying to use circle in a project [BusRaider](https://www.tindie.com/products/robdobson/busraider-graphics-wifi-debug-kit-for-rc2014/) and would like the main source code to in a peer-folder of circle (or similar). So that the source code of the main project is not inside the circle folder. I would then add circle as a submodule in the project. In other words a layout like this:

----projectFolder
----------circle
----------src

I also want to use REALTIME as I find the serial (UART) drops characters at higher speeds if I don't use it.

The advice for defining REALTIME is that it should be defined in circle/include/sysconfig.h but that requires changing the source inside of circle which would mean the submodule approach wouldn't work well.

I have tried to add a define into the code that runs make but that doesn't seem to work. For instance if I do:

cd circle/lib 
make CFLAGS=-DREALTIME=1

It fails to build as the CFLAGS define created seems to stop the CFLAGS += ... rules in the makefile from working

Is there a way that I can make this work?

hpingel, 2020-09-01:
Yes, create a file Config.mk in folder circle and put in the line
DEFINE += -DREALTIME

See README for more infos regarding Config.mk.

Don't change sysconfig.h directly. (Remark: If REALTIME is on you can't use WLAN)

robdobsn, 2020-09-01:
Thanks @hpingel, that's helpful. Can I create the Config.mk file in a folder outside the circle folder? I assume if I create it in the circle folder then it would be treated as a change to the submodule?

rsta2, 2020-09-01:
That's the way to do it, what @hpingel suggested. The file *Config.mk* is ignored by using *.gitignore*, so there shouldn't be a problem with it.

robdobsn, 2020-09-01:
I've tried to follow this advice but if I add Config.mk to the project as you suggest then cloning the project to an empty folder will result in there being no Config.mk file and then REALTIME will not be defined so the overall program will fail to work as it needs REALTIME defined in order to work correctly. Is there no way to define REALTIME outside the circle project so that I can treat the project as a self-contained submodule and ensure that it will build with the required flags when embedded in a project?

hpingel, 2020-09-01:
OK, it's only the question of the workflow. In the minisynth example Rene has created a configure script that dynamically creates the Config.mk file in the right place.

So the steps would be:
1) Clone your repo with recursive submodules.
2) Have a manual step to create the Config.mk (maybe by calling a configure script)
3) make

I am using self-created Docker images to build because my setup is even more complicated than that. In the Dockerfile I have several flavours of Config.mk and copy those into the right place.

robdobsn, 2020-09-01:
Ok, that makes sense, I will just create a Config.mk and copy it into the right place in the build script. 
Thanks very much for the suggestions and help. 
I'll close now so it doesn't hang around on the unresolved issues list.

closed 2020-09-01 by robdobsn