rsta2/circle #131: Destructor of linuxdevice vs. IsChainBootEnabled

hpingel, 2020-09-06:
Hi Rene,

I'm trying to get chainboot to work with Sidekick64. A CVCHIQDevice is used and that inherits the destructor of CLinuxDevice I guess. In there it always fails:

```
CLinuxDevice::~CLinuxDevice (void)
{
	assert (0);
}
```

This means my chainboot is always stuck there... Will it get worse if I modify the destructor? ;-)

Regards,
Henning

rsta2, 2020-09-07:
Hi Henning, you may probably remove this assertion and the application should restart then, but I'm afraid if the started application is using VCHIQ too, this will not work, because VCHIQ cannot be restarted. The firmware and thus Circle do not provide this. VCHIQ can only be (re-)initialized after a real hardware reset.

hpingel, 2020-09-07:
OK, this is very good to know, thank you for clarifying! The described VCHIQ no-restartable behaviour would not only apply to the ChainBoot scenario but also to the normal ShutdownMode "ShutdownReboot"/EXIT_REBOOT if I understand this correctly?

Is this already documented in you excellent documentation?

(This means I could consider to get one of my Smarthome power plugs into the scenario to power off and power on the Pi via a network command.)

hpingel, 2020-09-07:
Oh, the penny just dropped - why the http bootserver is called bootserver. Because it is designed to run *before* VCHIQ is potentially started. What I was trying to do is to use your webserver example for a multi-purpose webserver during the whole runtime of the application. The bootserver's functionality should be mixed into this as it will only work on its own!

rsta2, 2020-09-07:
First returning "ShutdownReboot"/EXIT_REBOOT should work in any case, if EnableChainBoot() has not been called before. In this case a real hardware reset is triggered using the watchdog device. Please note that the main.cpp file in sample/38-bootloader is slightly different from that of most other samples. If you want a real hardware reset reboot, you should use e.g. sample/01-gpiosimple/main.cpp.

Maybe the name "...bootserver" is a bit misleading here. With "chain boot" the loaded kernel image is copied over the old one in memory and then the entry point (0x8000 (AArch32) or 0x80000 (AArch64)) of the kernel image is entered. This works (even in user applications) when the old kernel image has shutdown everything it was using properly, but unfortunately not with all devices. It depends on the application.

hpingel, 2020-09-09:
> First returning "ShutdownReboot"/EXIT_REBOOT should work in any case

> With "chain boot" the loaded kernel image is copied over the old one in memory

Thank you for the clarification! This really answers all my questions!

closed 2020-09-09 by hpingel