rsta2/circle #132: Help on using the bootloader

yomboprime, 2020-09-12:
Hi,

My problem is that the bootloader works only once after copying the files to the SD. The uploading doesn't work the second time I try it.

The Config.mk file contains the following lines:
```
SERIALPORT = /dev/ttyUSB1
FLASHBAUD = 115200
USERBAUD = 115200
```
My compiler machine is a PC with Ubuntu.
My Pi is 1B model.

These are the steps I follow:

- Copy all the files from circle/boot to the SD except config.txt
- Power on the Pi. A kernel which shows a RGB image is run.
- Build the 02-screenpixel sample.
- ```make flash``` from there.
- The bootloader runs normally and I see in the USB to serial adapter that the transmitting led is blinking.
- After completion the sample program is run on the Pi. The Pi green LED blinks as it should, and the rectangle with a cross is shown on the screen.
- Then I try to make a change in the sample program, build again and ```make flash``` again.
- From the PC side all is normal and there are no errors, and the UART adapter led blinks.
- But from the Pi side, it doesn't seem affected. The Pi green LED continues to blink and the cross and rectangle keep showing.

Expected result: In the last step, the Pi should reboot and execute the program with the latest changes.

I have followed all the instructions like building the libraries with ```./makeall``` and building the bootloader with ```make``` in ```/boot```.

Thanks in advance.

rsta2, 2020-09-13:
What you have done, seems to be all right. But the Raspberry Pi has to be powered off after each try and powered on before the next one, so that the bootloader can take control again. Once a program has been started by the bootloader, the bootloader does not have control any more over the RPi and cannot automatically reboot the system. This has to be done manually.

yomboprime, 2020-09-13:
Thanks! Ahhh now I see. It works now. So the rainbow RGB image is the bootloader executing.

Should I make a PR with a note about this in bootloader.txt?

rsta2, 2020-09-13:
The rainbow image is created by the firmware after a hard (re-)boot and is a sign, that the bootloader is running, if it is installed on the SD card. The bootloader itself does not touch the screen.

Yes, please send a PR, if you want to. Please branch off from the *develop* branch, where development takes place and be sure that the author information in your commit is valid.

yomboprime, 2020-09-13:
Thanks. I can also restart the RPi from code under some event, instead of having to unplug power.

rsta2, 2020-09-14:
Yes, but this does only work with the help of the running test program. The common way to do that, is to include the class *CSerialDevice* with interrupt support in the program and to register a "magic received handler" using CSerialDevice::RegisterMagicReceivedHandler(), which calls reboot(), when the magic string is received. One can define this magic string for instance in *Config.mk* like that:

```
REBOOTMAGIC = MYMAGIC
```

Then when you enter "make flash" the RPi should be rebooted and restarted automatically with a new kernel image.

yomboprime, 2020-09-14:
Thanks, that is a better way of solving it.

PR #133 (2020-09-14): Add note about restarting RPi to re-run bootloader
Fixes #132
closed 2020-09-14 by yomboprime