rsta2/circle #136: UART drops characters without REALTIME and USB Keyboards don't work with REALTIME

robdobsn, 2020-09-19:
I'm using REALTIME as I can't get reliable performance from UART serial communications without it. Maybe there is something else (other than setting REALTIME) that can be done to make UART reliable? The UART FIFO is 16 bytes so even at 921600 baud (which I am using) there should be 16uS or so to respond to UART interrupts? But characters are dropped if REALTIME isn't set.

If using REALTIME then lines 158..165 in usbdevice.cpp result in USB keyboards failing to work. I have found that commenting out these lines results in USB keyboards actually working so I suspect the m_Speed check isn't correct - the value of m_Speed with a USB keyboard attached is 0 - USBSpeedLow.

rsta2, 2020-09-19:
Do you use the FIQ for the UART? You have to set the second parameter of the constructor of CSerialDevice to TRUE to do that. The UART works much better with the FIQ.

USB keyboards are normally low-speed devices, that's true. The USB driver disables the IRQ to up to 1ms, low-speed devices are used, not in any case, but sometimes. That's why REALTIME forbids using low-speed devices.

robdobsn, 2020-09-20:
I had planned to use the FIQ for something else if possible. What is it that REALTIME does that makes CSerialDevice drop characters? I can't see anything that would cause that to happen - it seems to happen even if I don't use USB.

rsta2, 2020-09-20:
A spin lock is used in CSerialDevice (and CScreenDevice) to ensure, that the output is not scrambled, if multiple cores try to write a message to the logger at the same time, if one of these classes is used as logging target. This would happen too, when a message is written from TASK_LEVEL and and IRQ will be triggered at this time and the IRQ handler writes a logging message too (which is allowed).

To prevent this, the IRQ is normally disabled, when the spin lock is acquired. Normally the serial output will not take that long, because the data is only written to a buffer in memory, where it is read by the serial IRQ handler. This is not the case for screen output. If the REALTIME option is defined, the IRQ is not disabled in CSpinLock::Acquire() any more.

I'm not satisfied with this solution any more and I'm looking for a better one, but it will take some time.

rsta2, 2020-09-20:
After thinking about this problem, I have no real explanation, why the UART is not reliable (on the receive direction) without REALTIME defined, when USB is not used. Do you write to the screen (class CScreenDevice) or access the SD card (class CEMMCDevice) while receiving data on the UART by any chance. Which RPi model are you using?

rsta2, 2020-09-21:
I tested UART receive on my own at 921600 baud on a RPi 3B, with these results:

| Interrupt | REALTIME | Screen scrolls | Result |
| --- | --- | --- | --- |
| FIQ | don't care | don't care | success |
| IRQ | defined | don't care | success |
| IRQ | not defined | no | success |
| IRQ | not defined | yes | failure |

These results were expected. REALTIME reduces the IRQ latency and disables device drivers or functions, which are not suitable for real-time operation. Such function is the concurrent access to the screen from TASK_LEVEL and IRQ_LEVEL.

There is an update on the *develop* branch (will be part of the next release), which allows to enable the system option USE_USB_SOF_INTR together with REALTIME now. With USE_USB_SOF_INTR defined, USB low- and full-speed devices can be used, even if REALTIME is defined.

So please use the code from the *develop* branch and define both REALTIME and USE_USB_SOF_INTR. This should resolve all your problems and the FIQ remains free for other tasks.

rsta2, 2020-10-02:
Circle Step43 has been released. One can enable the REALTIME option now to reduce IRQ latency and use USB keyboards at the same time too. There is a *sample/40-irqlatency* included, which allows to measure the actual maximum IRQ latency.

closed 2020-10-02 by rsta2

robdobsn, 2020-10-02:
Hi Rene

That's amazing!

Thank you so much

Rob