rsta2/circle #137: Moving from circle to circle-stdlib and keep on using net/httpdaemon.cpp

hpingel, 2020-09-22:
Hi,

I would like to discuss this issue with both of you ( @smuehlst and @rsta2 ). It's about the learning curve. I come from circle and I adapt to circle-stdlib so that I can use an httpclient with TLS. Works very nicely for me and was easy to learn and understand because there is nearly the drop-in replacement of circle's httpclient.cpp in circle-stdlib/src/circle-mbedtls/httpclient.cpp that allows me to do HTTPS.

On the other hand I am playing with the webserver/bootserver example of circle where TLS/HTTPS is not important at the moment (but maybe important in the future). In this case I am relying on circle's net/httpdaemon.cpp. For this case I didn't find a drop-in replacement in circle-stdlib which allows a reuse of existing code when moving from circle to circle-stdlib. My solution at the moment is to copy net/httpdaemon.cpp into my project and patch it for the use with circle-stdlib - which only means changing a few lines of code.

My perspective is on the learning user who is starting off with examples from circle. I know that circle-stdlib has its own nice examples and there are at least two HTTPS-webserver examples derived from mbedtls. I just want to suggest if the net/httpdaemon.cpp could be added to circle-stdlib or the httpdaemon.cpp in circle could be modified to compile with circle-stdlib (I got "ambiguous" error messages because of THTTPState - have to look it up - I couldn't mix includes from the net sections of both projects.). This would smoothen the learning curve when traversing to circle-stdlib while TLS usage for the webserver is not a priority. I know it would be strange to have a non-encrypting httpdaemon living in a subfolder indicating encryption (circle-stdlib/src/circle-mbedtls/).

This is how I patched the httpdaemon of circle to use it with circle-stdlib:

`--- /httpdaemon.circle.h
+++ /httpdaemon.tweaked.h
@@ -22,9 +22,13 @@
 
 #include <circle/sched/task.h>
 #include <circle/net/netsubsystem.h>
-#include <circle/net/http.h>
+#include <circle-mbedtls/http.h>
 #include <circle/net/socket.h>
 #include <circle/types.h>
+
+#define HTTP_MAX_MULTIPART_BOUNDARY 100
+
+using namespace CircleMbedTLS;
 
 class CHTTPDaemon : public CTask
 {
`

(I will paste another patch here in a minute)

rsta2, 2020-09-22:
You can easily use the class CHTTPDaemon by including its original header at `<circle/net/httpdaemon.h>`, even in circle-stdlib applications and this is the recommended way. You do not need to copy the source code from Circle. I could provide a modified sample program (from sample/21-webserver) to show this. I think, the only reason to directly include the HTTP daemon class into circle-stdlib would be to support TLS, but I guess there is no real demand for it and currently not the time to realize this.

Also if you really want to use your bare metal RPi as a server with TLS, you need a server certificate and to be able to create one, your RPi must be reachable via a unique host name on the Internet. I don't know, if anybody is using his RPi in that way. Without a dedicated server certificate, you will get error messages from the web browser, when trying to access the server. This makes the whole TLS server stuff less important.

hpingel, 2020-09-22:
From the certificate side I am prepared (I have a domain with Let's Encrypt wildcard cert that are checked via DNS challenge - so I have an endless amount of subdomain names at hand that I could throw at my Raspberry Pi collection), but in this case I just don't need HTTPS for the webserver at the moment. I just want to create a simple web interface and test how it works with the Sidekick application.

My problem is that at the same time I want to use mbedtls for the httpclient to do GET requests via HTTPS. This means that I need to include flies from circle-stdlib/mbedtls that will provide THTTPStatus and conflict with the THTTPStatus implementation in circle. Maybe it's just the way I do includes. Maybe with a smarter approach what to include where I don't get into this issue.

Basically, it comes down to either #include <circle/net/http.h> or #include <circle-mbedtls/http.h>. I can't do both.

rsta2, 2020-09-22:
I tried this here and I can include both `<circle/net/http.h>` and `<circle-mbedtls/http.h>` in the same file (before the `using namespace` directive). Then you have one type `THTTPStatus` with global scope and one which is valid in namespace CircleMbedTLS only. You must explicitly use the prefix `CircleMbedTLS::` to access definitions of `CircleMbedTLS::THTTPStatus` or `::THTTPStatus` for the other global one.

hpingel, 2020-09-22:
Thank you so much for your help here! You are right! Using standard namespace via :: resolves all the issues!

(P.S.: @rsta2: A friendly fellow from the retro community at Forum64 also involved with the Sidekick project has asked me to ask you if you were involved in projects around TFPCX in the past? He said that your name sounds familiar to him.)

closed 2020-09-23 by hpingel