rsta2/circle #139: Q: Benchmark

actonDev, 2020-10-05:
Not an issue, but I'm not aware of any community for this.

Are there any (even rough) benchmark off a sample application performance?
That is running the sampe on full os (raspbian, tinycore) and with circle bare metal.
Also about boot time.

I'm wondering (at least in my case) if what can one gain going bare metal.

rsta2, 2020-10-06:
Boot time depends very much on which devices have to be used. My [Unitest](https://github.com/rsta2/unitest) program, which is not optimized for boot time, booted in about 11 seconds to prompt in a quick test on a RPi 3B+ (32-bit mode). There is included firmware boot (2 seconds), some optional Act LED blinking (3), optional delay loop calibrating (0.66), power on delay for two USB hubs (2*0.5), optional delay for waiting on DHCP to setup (2). So overall without the options it would take somewhat over 5 seconds. The program supports multiple cores, USB and TCP/IP.

I'm not aware of any performance benchmarks, but boot time and performance are not the most important reasons for using bare metal solutions. Often it's interrupt latency, that you have more control, what the system is doing and sometime the simple fact, that you can power off the system at any time without data loss, as long you are not accessing the SD card.

actonDev, 2020-10-06:
Thanks for these answers!
Also helped clear some misconceptions about bare metal

closed 2020-10-06 by actonDev