rsta2/circle #140: Reading GPIO

yomboprime, 2020-10-08:
In my application I need to sample a 16 bit bus that has a 4 MHz signal. I use the ```CGPIOPin::ReadAll()``` function to read all the pins.

The docs say:
```
The GPIO block of the Raspberry Pi is able to [...] input data with up to about 15 MHz. The exact rates
depend on the Raspberry Pi model and the system clock.
```

I've made a simple program (kernel.cpp is attached) to benchmark the GPIO read speed, these are the results with my Pi's:
```

Raspberry Pi 1B: 9.3 MSamples/s
Raspberry Pi 4B (32 bits): 5.7 MSamples/s
Raspberry Pi 4B (64 bits): 5.7 MSamples/s
Raspberry Pi 4B (32 bits, defined REALTIME): 5.8 MSamples/s

```
So the Pi1 is faster than Pi4. I need at least 8 MSamples/s (by Nyquist law, ideally more). But the Pi1 only has 1 cpu wich will not be able to process the signal and do other tasks I need. The Pi4 would be ideal with 4 fast cpus, but it seems it can't sample at sufficient rate.

So my question is, can I do something to achieve > 10 MHz input rate on the Pi4?

[kernel.cpp.txt](https://github.com/rsta2/circle/files/5348265/kernel.cpp.txt)
[kernel.h.txt](https://github.com/rsta2/circle/files/5348266/kernel.h.txt)

rsta2, 2020-10-08:
By default the Raspberry Pi 4B runs with a CPU clock of 600 MHz only after boot. For full speed operation (1500 MHz) you have to use the class *CCPUThrottle* as follows:

kernel.h:

```
#include <circle/cputhrottle.h>

...
class CKernel
{
...
private:
	// do not change this order
	CMemorySystem		m_Memory;
	CActLED		        m_ActLED;
	CKernelOptions		m_Options;
	CDeviceNameService	m_DeviceNameService;
	CCPUThrottle		m_CPUThrottle;
	CScreenDevice		m_Screen;
...
};
```

kernel.cpp:

```
CKernel::CKernel (void)
:	m_CPUThrottle (CPUSpeedMaximum),
	m_Screen (m_Options.GetWidth (), m_Options.GetHeight ()),
...
```

In a quick test I reached 14.5 MSamples/s with your program, modified this way. Please note that the SoC may get hot, when running all the time at 1500 MHz. The firmware automatically throttles the CPU, when it comes near 85 °C. You may need a fan, if it runs longer at this speed.

yomboprime, 2020-10-08:
Thanks! Il try it tonight. I already have a fan in my Pi4 :-)

yomboprime, 2020-10-09:
It works well, as expected :-)

rsta2, 2020-11-14:
I think, this can be closed.

closed 2020-11-14 by rsta2