rsta2/circle #141: Some questions re: MCP23017 and interrupts

dwhinham, 2020-10-08:
Hi Rene!

Not a bug report, but more like some questions about how I might best design something using Circle.

Background: I'm wanting to build a control surface for my Pi3+ Circle-based project. The control surface will have a rotary encoder with push button, and 10 more buttons to control various functions. The project uses FIQ interrupts for the UART (I'm using this for MIDI input) and renders audio inside the `GetChunk()` callback of the audio driver. The "main loop" updates an I2C-connected OLED display every 16ms (~60Hz).

To save on GPIO usage for the control surface, one idea was to use a port expander. The MCP23017 is a popular choice which interfaces via I2C. This device can be polled (which of course takes CPU time and uses the I2C bus), or it can be wired (and configured) to send an interrupt when it sees any of its inputs change. When an interrupt is received, you know that you need to read the port expander's registers via I2C and get the latest pin state.

My current prototyping setup has the two inputs of the rotary encoder wired to `GPA0` and `GPA1` of the expander. Code to read the encoder is based on a [state machine lookup table](http://www.buxtronix.net/2011/10/rotary-encoders-done-properly.html) for its simplicity and debouncing properties. Turning the encoder increases/decreases a value, and if that value changes, a message is logged to the console.

At first I tried the polling method. It seems like there's too much going on in my kernel, because encoder state transitions are rarely detected in time unless I turn the encoder extremely slowly, even between "clicks".

After simplifying my kernel and disabling everything except encoder reading, things are better - not perfect, but it behaves more like I expect and we "catch" most of the encoder events.

So my next step was to try interrupt - the INTA pin is now wired to a GPIO which responds to a falling-edge ISR. I naively tried to do the I2C `Read()` operation inside the ISR, but I would start to get freezes - and I realised that doing this is probably illegal.

Instead, I set a flag in the ISR, and if the main loop sees that the flag has changed, it then does the I2C read. This also kind-of works if the main loop doesn't do much, but as soon as the kernel does more heavy lifting, we're back to the same problems as just polling and using interrupts hasn't solved everything.

At this point I'm thinking of a few options:

- Maybe I have to go multi-core (it was going to happen eventually anyway) and dedicate a core to servicing the port expander. I'd probably have to have some synchronisation to deal with (as there's also an LCD/OLED on the I2C bus).
- Maybe I can somehow have a high priority "input task" that waits, and the ISR "signals"/wakes that task when there's new stuff to read from the expander, so that it quickly takes over when the ISR returns.
- Maybe this idea isn't going to work too well and I should just use the MCP for buttons; giving the rotary encoder a pair of dedicated GPIOs instead.

If you have any other ideas I'd really appreciate any advice! I'm still figuring out this bare metal stuff as I go along (but I'm really enjoying Circle!) so any advice would be really appreciated.

Cheers!

rsta2, 2020-10-09:
Hi Dale, please modify the parameter TASK_LEVEL to IRQ_LEVEL in [this line](https://github.com/rsta2/circle/blob/master/lib/i2cmaster.cpp#L109) and go for a test with INTA again. By default `CI2CMaster::Read()` is callable from TASK_LEVEL only, but with this modification, you should be able to call it directly from the GPIO IRQ handler.

dwhinham, 2020-10-10:
Ahhh, nice and simple - that's got I2C `Read()` working inside the interrupt. Thankyou!

I still have a problem with getting the MCP to clear its interrupt status; if I just do one read of the MCP's GPIOA or INTCAPA register, the interrupt state will still be active, and my interrupt pin will still be held low.

I think it's probably because I read it so fast that another interrupt occurs _while_ reading (because of contact bounce within the encoder, perhaps), and I need to deal with this somehow.

I'll try some stuff and let you know how I get on. 🙂

dwhinham, 2020-10-18:
Closing this as I seem to have interrupts working reliably for the MCP; I can just keep reading the MCP's INTCAP register within the ISR as while my interrupt GPIO pin is low. This way, the MCP doesn't get stuck in in the interrupt condition.

Thanks again for the I2C IRQ level hint!

closed 2020-10-18 by dwhinham