rsta2/circle #142: USB mouse wheel support

hinxx, 2020-10-16:
First of all, I would like to share some praise for the effort and the state of this project! It is amazing! I like the amount of supported features and the simple coding that is used.
I'm interested in using USB mouse wheel. Looking at the and `usbhiddevice.cpp` it seems that boot protocol is in use. My USB HID knowledge is not great, but reading up on USB HID specs 1.11 it seems that wheel information would not be part of information obtained (at least not required by the standard). Then, the report protocol is mentioned in one header, but not used anywhere in the code, AFAICT; this being something that would allow us to get wheel info. I might be wrong here, let me know, please.

Would the addition of mouse wheel handling to the current code base involve huge effort? I'll be investigating this on my side, and are willing to do some coding to support this, so any pointers to the direction this should go are welcomed.

Keep up the good work!!

rsta2, 2020-10-16:
Thank you first! Honestly I don't know, how the wheel info can be captured. I did a quick attempt to find it out earlier, but without success. I checked two mice and both provide the USB interface 3-1-2 (HID class, boot subclass, mouse protocol) only. I don't know, if a HID report descriptor can be requested from these devices and if the report protocol can be enabled, even considered that the subclass is "boot".

This requires some research. If you want to do this, have a look at [this driver](https://github.com/rsta2/circle/blob/master/lib/usb/usbgamepadstandard.cpp), which contains a parser for the report descriptor of a HID gamepad device. The USB mouse driver itself is [here](https://github.com/rsta2/circle/blob/master/lib/usb/usbmouse.cpp).

If you can find it out, it would be great, if you would share this information and/or your code. Maybe we would have to talk about required modifications to the mouse API, which is declared [here](https://github.com/rsta2/circle/blob/master/include/circle/input/mouse.h) (raw status handler) and [here](https://github.com/rsta2/circle/blob/master/include/circle/input/mousebehaviour.h) (event handler).

hinxx, 2020-10-18:
Thanks for the pointers. I managed to get something working - see PR #144.
Consider it an initial attempt that needs cleaning and further testing, but all in all it works for me with three different mice.

closed 2020-11-24 by rsta2