rsta2/circle #145: Mouse and vc4 interface

hinxx, 2020-10-19:
Has anyone used a mouse with the vc4 graphics and noticed the displacement of the mouse cursor vs. the mouse position?

I do not have a good grip on how the whole graphics matter is dealt with on RPi at this low level, it seems quite intricate and complex with steep learning curve.

In my GLES app I see that the the mouse click is registered 48 pixels to the left, and 48 pixels to up, from where the mouse cursor is drawn on screen. Looking a bit closer it seems that the framebuffer size as detected by `CScreenDevice` is not 1920x1080 but 96 pixels shy in both directions; at 1824x984 pixels. At least this is what I see with my RPi zero W.

OTOH, the dispmax interface seems to think that the window/surface handed over to eGl is 1920x1080 as returned by the `vc_dispmanx_display_get_info`. Looking at the screen the graphics look fine, occupying complete screen/monitor, but the mouse cursor is acting by other rules. The mouse position is reported from 0,0 to 1919,1079 which seem OK to me, with the 1919,1079 being off screen bottom right. Again, the mouse clicks reported are in the "right" place, just the cursor is drawn at and offset.

I will try to use cursors as seen in SDL http://hg.libsdl.org/SDL/file/1055291959fd/src/video/raspberry/SDL_rpimouse.c#l64 to see if it helps.

rsta2, 2020-10-19:
Haven't noticed this yet. What happens, when you explicitly set the screen size to 1920x1080 in the constructor of m_Screen?

hinxx, 2020-10-19:
The framebuffer now reports the size 1920x1080.
The mouse cursor is still displaced compared to clicks. Horizontally for (what it appears) about twice as much, 96 pixels, on both sides. Vertically about the same same as before, 48 pixels. 
Also the mouse cursor now does *not* travel off screen anymore but is bound on all 4 sides with the mentioned amount of pixels.
At the same time the mouse travel is still from 0,0 to 1919,1079.

rsta2, 2020-10-19:
Unfortunately I cannot reproduce this here. I modified the sample hello_triangle2, that it shows a mouse cursor and displays its coordinates from the mouse event handler via the serial interface. On a RPi Zero W I can move the mouse cursor in all corners and the coordinates and cursor display is as expected. Also the screen size is automatically detected right in CScreenDevice. Maybe it has something to do with the used display hardware?

hinxx, 2020-10-19:
This is what is see as maximum extents of the physical screen with 10-usbmouse sample.
[image]

I will try another monitor and also give it a run with a modified triangle2 sample as well!

Thanks for confirming that it should work!!

hinxx, 2020-10-19:
>  I modified the sample hello_triangle2, that it shows a mouse cursor and displays its coordinates from the mouse event handler via the serial interface

@rsta2 , do you mind sharing your modified hello_triangle2 code?
I'm not seeing any difference on my side if I use another monitor. BTW, my monitors use 16:9 1080p @60Hz if it matters.

rsta2, 2020-10-19:
Do you have any settings in *config.txt*, which may affect the screen (e.g. overscan)? On my relatively old 1280x1024 (at 60 Hz) monitor there is no border. The image fills the whole display area.

The test program is appended. It's a quick hack. You need a file *cmdline.txt* with this contents on the SD card:

```
logdev=ttyS1
```

An terminal at 115200 Bps must be attached to the serial interface to see the output (e.g. `make monitor`). I moved the mouse into all corners and the coordinates are the expected ones (0/0, 0/1023, 1279/0, 1279/1023). The hotspot of the mouse cursor is exactly in the corner. I use the firmware from Aug 06 2020. It can be downloaded in the *boot/* folder using `make`.

[hello_triangle2_mousetest.zip](https://github.com/rsta2/circle/files/5404266/hello_triangle2_mousetest.zip)

hinxx, 2020-10-19:
Thanks for the code, that should make the test less prone to any of my potential coding errors..

I ran the test and my mouse still does not move all the way into the corner; see the image.

[image]

My screen get detected as 1824x984, not 1920x1080. Mouse motion is within these numbers: 0,0 to 1823,983.

I have no config.txt on my sd card and the firmware is from the ./boot folder.

In my case the image is also filling up the complete screen, as you report, only the mouse does not move all the way into the corners.

In the vc4 test case the image area is bigger than the screen framebuffer area; I can tell by seeing the indented boot messages, when not using the serial console. Another telltale sign that screen framebuffer is smaller than vc4 area is the location of the rotor as seen in the 10-usbmouse test earlier.

Maybe I need to add some parameters to the config.txt to get larger framebuffer. Nevertheless, this does not seem to affect the vc4 area size.

hinxx, 2020-10-19:
> Do you have any settings in config.txt, which may affect the screen (e.g. overscan)?

Using:
```
disable_overscan=1
```
in `config.txt` solves the issue for me.

closed 2020-10-19 by hinxx

hinxx, 2020-10-20:
BTW, thank you for your time in helping out!!!