rsta2/circle #146: assertion failed: m_pURB == pURB

hinxx, 2020-10-23:
I'm attaching two cases of failures seen running sample/10-usbmouse sample.
This was provoked by removing / attaching some mice I have.

```
...
00:03:12.01 dwhci: Transaction failed (status 0x202)
00:03:12.51 dwhci: Transaction failed (status 0x202)
00:03:13.01 dwhci: Transaction failed (status 0x202)
00:03:13.51 dwhci: Transaction failed (status 0x202)
00:03:13.86 kernel: Mouse removed
00:03:13.86 kernel: Please attach an USB mouse!
00:03:13.87 usbdev0-1: Device ven46d-c539 removed
00:03:36.25 usbhiddevice.cpp(221): stack[11] is 0x140B0
00:03:36.25 usbhiddevice.cpp(221): stack[13] is 0xB3FC
00:03:36.25 usbhiddevice.cpp(221): stack[17] is 0xB654
00:03:36.25 usbhiddevice.cpp(221): stack[20] is 0xBBF4
00:03:36.25 usbhiddevice.cpp(221): stack[23] is 0xB314
00:03:36.25 usbhiddevice.cpp(221): stack[35] is 0xBAD0
00:03:36.25 usbhiddevice.cpp(221): stack[47] is 0xBBF4
00:03:36.25 usbhiddevice.cpp(221): stack[51] is 0x24688
00:03:36.25 usbhiddevice.cpp(221): stack[53] is 0x24748
00:03:36.25 usbhiddevice.cpp(221): stack[54] is 0x1BF44
00:03:36.25 usbhiddevice.cpp(221): stack[63] is 0x2478C
00:03:36.25 usbhiddevice.cpp(221): assertion failed: m_pURB == pURB
```

```
...
00:00:42.64 dwhci: Transaction failed (status 0x202)
00:00:42.64 dwhci: Transaction failed (status 0x202)
00:00:42.64 dwhci: Transaction failed (status 0x202)
00:00:42.64 kernel: Mouse removed
00:00:42.64 kernel: Please attach an USB mouse!
00:00:42.64 usbdev0-1: Device ven46d-c539 removed
00:00:42.64 usbhiddevice.cpp(228): stack[11] is 0x14144
00:00:42.64 usbhiddevice.cpp(228): stack[13] is 0xB3FC
00:00:42.64 usbhiddevice.cpp(228): stack[17] is 0xB654
00:00:42.64 usbhiddevice.cpp(228): stack[20] is 0xBBF4
00:00:42.64 usbhiddevice.cpp(228): stack[23] is 0xB314
00:00:42.64 usbhiddevice.cpp(228): stack[35] is 0xBAD0
00:00:42.64 usbhiddevice.cpp(228): stack[47] is 0xBBF4
00:00:42.64 usbhiddevice.cpp(228): stack[51] is 0x24730
00:00:42.64 usbhiddevice.cpp(228): stack[53] is 0x247F0
00:00:42.64 usbhiddevice.cpp(228): stack[54] is 0x1BF44
00:00:42.64 usbhiddevice.cpp(228): stack[63] is 0x24834
00:00:42.64 usbhiddevice.cpp(228): assertion failed: m_pURB == pURB

```

Seems both cases were provoked by removing device `ven46d-c539`.
The same device is very prone to make `dwhci: Transaction failed (status 0x202)` appear.

rsta2, 2020-10-24:
This seems to be a bug, caused by the recent modifications for USB plug-and-play support. I will try to reproduce it. These "Transaction failed (status 0x202)" messages can occur, when a device is removed.

hinxx, 2020-10-24:
> These "Transaction failed (status 0x202)" messages can occur, when a device is removed.

For this particular mouse only, I see them all the time, as soon as the the device is connected, at a quite frequent rate, ~20 per second or so.

rsta2, 2020-10-24:
I know, what the problem for the initial issue is. There would be a relatively simple fix for it, but that device, which generates the 0x202 transaction errors, wouldn't work at all with it any more. These messages are another issue. They may happen, when the device is removed, but not all the time, when the device is attached.

hinxx, 2020-10-24:
If I understand correctly, the `Transaction failed (status 0x202)` messages should not be appearing during normal use. I assume this might be due to the device issues, since they do appear. Then, the fix to `assertion failed: m_pURB == pURB` would mean that such device would not be working anymore, at all.

rsta2, 2020-10-24:
I will need more time to check this. I cannot tell at the moment, if these `Transaction failed (status 0x202)` messages are caused by a device firmware issue. There are devices, which do not work, even with the xHCI controller of the RPi 4, which is doing much more in its own firmware, but on the RPi 1-3 and Zero one can do much more wrong in the USB stack, which is running on the ARM CPU.

hinxx, 2020-10-24:
Take tour time. It is not a big thing for me. Let me know if I can help out in any way!

rsta2, 2020-10-25:
The issue should be fixed on the *develop* branch. It would be very helpful if you could test it again. I will merge the PR #144 soon to this branch, so that you can test all together.

hinxx, 2020-10-26:
While testing your fix I still encountered the issue twice, with two different mice.

```
00:04:41.29 kernel: Mouse removed
00:04:41.29 kernel: Please attach an USB mouse!
00:04:41.30 usbdev0-1: Device ven46d-c539 removed
00:04:44.85 usbhiddevice.cpp(227): stack[9] is 0x13E30
00:04:44.85 usbhiddevice.cpp(227): stack[17] is 0xB7D4
00:04:44.85 usbhiddevice.cpp(227): stack[19] is 0xE574
00:04:44.85 usbhiddevice.cpp(227): stack[21] is 0x237C4
00:04:44.85 usbhiddevice.cpp(227): stack[27] is 0x1DC3C
00:04:44.85 usbhiddevice.cpp(227): stack[37] is 0xBD0C
00:04:44.85 usbhiddevice.cpp(227): stack[49] is 0xBDC4
00:04:44.85 usbhiddevice.cpp(227): stack[53] is 0x249D8
00:04:44.85 usbhiddevice.cpp(227): stack[55] is 0x24A98
00:04:44.85 usbhiddevice.cpp(227): stack[56] is 0x1BF44
00:04:44.85 usbhiddevice.cpp(227): assertion failed: m_pURB == pURB
```

```
00:01:54.43 kernel: Mouse removed
00:01:54.43 kernel: Please attach an USB mouse!
00:01:54.44 usbdev0-1: Device ven2318-2808 removed
00:01:58.30 usbhiddevice.cpp(227): stack[9] is 0x13E30
00:01:58.30 usbhiddevice.cpp(227): stack[17] is 0xB7D4
00:01:58.30 usbhiddevice.cpp(227): stack[19] is 0xE574
00:01:58.30 usbhiddevice.cpp(227): stack[21] is 0x237C4
00:01:58.30 usbhiddevice.cpp(227): stack[27] is 0x1DC3C
00:01:58.30 usbhiddevice.cpp(227): stack[37] is 0xBD0C
00:01:58.30 usbhiddevice.cpp(227): stack[49] is 0xBDC4
00:01:58.30 usbhiddevice.cpp(227): stack[53] is 0x249D8
00:01:58.30 usbhiddevice.cpp(227): stack[55] is 0x24A98
00:01:58.30 usbhiddevice.cpp(227): stack[56] is 0x1BF44
00:01:58.30 usbhiddevice.cpp(227): assertion failed: m_pURB == pURB

```

hinxx, 2020-10-26:
I took the liberty to address the `Transaction failed (status 0x202)` issue in a separate issue, #149, since it still persists even with the above fixed.

rsta2, 2020-10-26:
There is a new approach to fix this issue on the *usb-pnp-fix* branch. Can you please test this again. Thanks! Unfortunately I wasn't able to reproduce the problem here, so I need your help.

hinxx, 2020-10-26:
Using the code in usb-pnp-fix branch it did not fail with the same as above, for at least 128 times  :).. but then this appeared:

```
00:06:21.96 kernel: Mouse removed
00:06:21.96 kernel: Please attach an USB mouse!
00:06:21.96 usbdev0-1: Device ven248a-8367 removed
00:06:23.85 usbdev0-1: Device ven248a-8367 found
00:06:23.85 usbdev0-1: Interface int3-1-2 found
00:06:23.86 usbdev0-1: Using device/interface int3-1-2
00:06:23.86 usbdev0-1: Interface int3-1-1 found
00:06:23.87 usbdev0-1: Using device/interface int3-1-1
00:06:23.93 CUSBRequest: Trying to allocate more than 128 instances
```

rsta2, 2020-10-26:
Did you really attached the mouse 128 times? :) I suppose, this will not happen in a normal use case. Thanks for testing.

hinxx, 2020-10-26:
Yeah, 128 times I guess.. :blush: 
No worries, it does not seem like something a normal user would do.

Otherwise, no issues with the code in the usb-pnp-fix branch, so far.

rsta2, 2020-10-26:
That's a lot. Thanks again for testing that carefully. I will think about again, if this limit can be removed somehow. It's not really nice, but unfortunately not easy to do.

rsta2, 2020-10-26:
There is some good chance, that the limit of USB device attachments / removes is gone with the latest update on the *usb-pnp-fix* branch. Please can you give it another test. To safe time, you can reduce the maximum number of `CUSBRequest` instances [here](https://github.com/rsta2/circle/blob/usb-pnp-fix/lib/usb/dwhcidevice.cpp#L123) (2nd parameter, e.g. set it to 4). I have reverted the parameter to 32 now.

hinxx, 2020-10-28:
@rsta2, you've done a great job!
After no less than 10 minutes of brutal USB connector abuse and 310 mouse removals later, no errors were observed. Love it!

rsta2, 2020-10-28:
Thank you very much @hinxx, also for stressing your USB connector that much. I hope it is still working. ;) I merged the branch *usb-pnp-fix* into *develop* now and removed *usb-pnp-fix*.

closed 2020-10-31 by hinxx