rsta2/circle #147: CDWHCITransferStageData Trying to allocate more than 16 instances

hinxx, 2020-10-23:
Juggling the mice while running sample/10-usbmouse causes the following to be printed in serial console:
```
00:01:21.31 CDWHCITransferStageData: Trying to allocate more than 16 instances
```
Afterwards the app becomes unresponsive.

rsta2, 2020-10-24:
Have you plug-and-play removed and re-attached the mice, before this happens?

hinxx, 2020-10-24:
Yes, I was doing this quite a lot. I would think about 16 times, as the number in the message says. 15 times with no issue, and the message only appeared at the last re-attachment.

rsta2, 2020-10-24:
I think, I know the reason for the problem and I can fix it. Will do so soon.

rsta2, 2020-10-25:
The issue has been fixed on the *develop* branch.

hinxx, 2020-10-26:
After testing this it looks like the problem is solved; I do not get `CDWHCITransferStageData: Trying to allocate more than 16 instances` message any more, even after reconnecting the mouse >16 times.

hinxx, 2020-10-26:
There seems to be a regression introduced compared to the master branch.
If I remove/re-attach the mouse quickly the `10-usbmouse` sample from the develop branch locks up and a reboot is required.
This is not the case with the master branch.

rsta2, 2020-10-26:
Thanks for testing. I'm afraid, I will not be able to handle very quick USB device attachments / removes in a row. With the latest fix from issue #146 it may be better, but not perfect.

hinxx, 2020-10-26:
OK, I have not big issues with the fact that super fast re-attachments may not work.

closed 2020-10-31 by hinxx