rsta2/circle #148: USB Keyboard Raw data issue

XORElectronics, 2020-10-23:
Hello. I try to play a bit with the USB Keyboard function and here especially in the RAW mode. 
While it works well mostly i get wrong data ( 0x01 0x01 0x01.... for all) in different key combinations. So if i press 3 (or more) keys like Z,X,C it will set all key data to 0x01 until i release one key again. Other key combinations work well up to 6 keys.
They are different combinations with a different amount of pressed keys which have this problem.
I tested 2 different USB keyboards and both show the issue (though both with different key combinations).
Running on a Pi-Zero, Compiling with Gnu toolchain 9.2-2019-12. Keyboard directly connected, no Hub.

rsta2, 2020-10-24:
Some USB keyboards send this 0x01, 0x01, ... data, when too many keys are pressed in special combinations. This has to be filtered out by the keyboard report handler. Because 0x01 is not a valid key code, this should not be a problem. USB keyboards are usually not able to report all pressed keys in all combinations.

XORElectronics, 2020-10-24:
Oh thanks a lot and sorry for making that an issue!
 
25+ years in IT and i was not aware of that, haha
Always something new to learn. Thanks and big up for the great work!

rsta2, 2020-10-25:
No problem and thanks!

closed 2020-10-25 by rsta2