rsta2/circle #149: Transaction failed (status 0x202)

hinxx, 2020-10-26:
This error was was noticed in the #146 already, but I'm pulling it out into a separate issue in order not to mix the two (distinct) problems.

I just wanted to clarify when and how I get these messages with the Logitech G403 mouse (not a mouse/keyboard combo). Observed with current master and develop branch code (from today).

After 10-usbmouse sample start with the mouse attached I can see messages appear in 500ms intervals. I have not moved the mouse or pressed a button so far.

```

logger: Circle 43.1 started on Raspberry Pi Zero W
00:00:01.00 timer: SpeedFactor is 1.00
00:00:02.14 dwhci: No device connected to root port
00:00:02.14 kernel: Compile time: Oct 26 2020 09:49:47
00:00:02.15 kernel: Please attach an USB mouse, if not already done!
00:00:07.48 usbdev0-1: Device ven46d-c539 found
00:00:07.48 usbdev0-1: Interface int3-1-1 found
00:00:07.49 usbdev0-1: Using device/interface int3-1-1
00:00:07.49 usbdev0-1: Interface int3-1-2 found
00:00:07.50 usbdev0-1: Using device/interface int3-1-2
00:00:07.50 usbdev0-1: Interface int3-0-0 found
00:00:07.51 usbdev0-1: Using device/interface int3-0-0
00:00:07.58 dwroot: Device configured
00:00:07.58 kernel: USB mouse has 16 buttons
00:00:07.58 kernel: USB mouse has a wheel
00:00:09.01 dwhci: Transaction failed (status 0x202)
00:00:09.51 dwhci: Transaction failed (status 0x202)
00:00:10.01 dwhci: Transaction failed (status 0x202)
00:00:10.51 dwhci: Transaction failed (status 0x202)
00:00:11.01 dwhci: Transaction failed (status 0x202)
00:00:11.51 dwhci: Transaction failed (status 0x202)
00:00:12.01 dwhci: Transaction failed (status 0x202)
00:00:12.51 dwhci: Transaction failed (status 0x202)
00:00:13.01 dwhci: Transaction failed (status 0x202)
00:00:13.51 dwhci: Transaction failed (status 0x202)
00:00:14.01 dwhci: Transaction failed (status 0x202)
00:00:14.51 dwhci: Transaction failed (status 0x202)
00:00:15.01 dwhci: Transaction failed (status 0x202)
00:00:15.51 dwhci: Transaction failed (status 0x202)
00:00:16.01 dwhci: Transaction failed (status 0x202)
00:00:16.51 dwhci: Transaction failed (status 0x202)
00:00:17.01 dwhci: Transaction failed (status 0x202)
00:00:17.51 dwhci: Transaction failed (status 0x202)
00:00:18.01 dwhci: Transaction failed (status 0x202)
00:00:18.51 dwhci: Transaction failed (status 0x202)
00:00:19.01 dwhci: Transaction failed (status 0x202)
00:00:19.51 dwhci: Transaction failed (status 0x202)
00:00:20.01 dwhci: Transaction failed (status 0x202)
00:00:20.51 dwhci: Transaction failed (status 0x202)
00:00:21.01 dwhci: Transaction failed (status 0x202)
00:00:21.51 dwhci: Transaction failed (status 0x202)
00:00:22.01 dwhci: Transaction failed (status 0x202)
00:00:22.51 dwhci: Transaction failed (status 0x202)
00:00:23.01 dwhci: Transaction failed (status 0x202)
00:00:23.51 dwhci: Transaction failed (status 0x202)
00:00:24.01 dwhci: Transaction failed (status 0x202)
00:00:24.51 dwhci: Transaction failed (status 0x202)
00:00:25.01 dwhci: Transaction failed (status 0x202)
00:00:25.51 dwhci: Transaction failed (status 0x202)
00:00:26.01 dwhci: Transaction failed (status 0x202)
00:00:26.51 dwhci: Transaction failed (status 0x202)
00:00:27.01 dwhci: Transaction failed (status 0x202)
00:00:27.03 dwhci: Transaction failed (status 0x82)
00:00:27.03 dwhci: Transaction failed (status 0x82)
00:00:27.05 kernel: Mouse removed
00:00:27.05 kernel: Please attach an USB mouse!
00:00:27.06 usbdev0-1: Device ven46d-c539 removed

```

Once I move the mouse the rate at which the messages appear increases correspondingly to the mouse movement speed.

```
00:01:20.19 dwhci: Transaction failed (status 0x202)
00:01:20.20 dwhci: Transaction failed (status 0x202)
00:01:20.21 dwhci: Transaction failed (status 0x202)
00:01:20.22 dwhci: Transaction failed (status 0x202)
00:01:20.46 dwhci: Transaction failed (status 0x202)
00:01:20.46 dwhci: Transaction failed (status 0x202)
00:01:20.47 dwhci: Transaction failed (status 0x202)
00:01:20.48 dwhci: Transaction failed (status 0x202)
00:01:20.49 dwhci: Transaction failed (status 0x202)
00:01:20.50 dwhci: Transaction failed (status 0x202)
00:01:20.50 dwhci: Transaction failed (status 0x202)
00:01:20.51 dwhci: Transaction failed (status 0x202)
00:01:20.52 dwhci: Transaction failed (status 0x202)
00:01:20.53 dwhci: Transaction failed (status 0x202)
00:01:20.54 dwhci: Transaction failed (status 0x202)
00:01:20.54 dwhci: Transaction failed (status 0x202)
00:01:20.55 dwhci: Transaction failed (status 0x202)
00:01:20.63 dwhci: Transaction failed (status 0x202)
00:01:20.64 dwhci: Transaction failed (status 0x202)
00:01:20.65 dwhci: Transaction failed (status 0x202)
00:01:20.66 dwhci: Transaction failed (status 0x202)
00:01:20.67 dwhci: Transaction failed (status 0x202)
00:01:20.67 dwhci: Transaction failed (status 0x202)
00:01:20.68 dwhci: Transaction failed (status 0x202)
00:01:20.69 dwhci: Transaction failed (status 0x202)
00:01:20.70 dwhci: Transaction failed (status 0x202)
00:01:20.71 dwhci: Transaction failed (status 0x202)
00:01:20.71 dwhci: Transaction failed (status 0x202)
```

The same happens for each button press; one such message per button press.

rsta2, 2020-10-26:
According to [this](https://elinux.org/RPi_USB_Mouse_devices) the Logitech G400 Gaming Mouse does work on Linux too with problems on the Raspberry Pi. Please search for "G400" in this document. I guess the G403 may have similar problems. Perhaps I have to mark it as "not supported".

hinxx, 2020-10-26:
I was not aware of that page. In this case, this mouse can can indeed be marked as "not supported".

closed 2020-10-26 by hinxx

rsta2, 2020-10-27:
I have added this mouse to the device (in-) compatibility list in the project wiki.