rsta2/circle #150: Getting raw and text keyboard input

hinxx, 2020-10-26:
I would need to get both, raw and text inputs for my application.
Is there are special reason not to allow this, when looking at this part of the code in `lib/usb/usbkeyboard.cpp`?:

https://github.com/rsta2/circle/blob/175e6bca75e03e42337e5aac54b55e1fde7c3441/lib/usb/usbkeyboard.cpp#L162

I've commented out that `return;` and now my app is able to obtain both types of keyboard inputs, without any issues that I could notice.

rsta2, 2020-10-27:
I've added a "mixed mode" to the USB keyboard driver on the *develop* branch. It allows to use both modes (raw and cooked) together and will be enabled with:

```
pKeyboard->RegisterKeyStatusHandlerRaw (YourRawHandler, TRUE);
```

Simply removing this "return" didn't do it, because the LED handling in `UpdateLEDs()` is also involved.

hinxx, 2020-10-28:
I have no LEDs on the keyboard to notice any difference.
Anyways, I just tested your change and it works for me.
Thank you!

closed 2020-10-31 by hinxx