rsta2/circle #152: SPI master accessibility

hinxx, 2020-10-28:
I'm looking into having 2 SPI devices connected to the SPI0 master on RPi zero, each using one of the CE[0,1]# pins.
The plan is to provide two separate device support libraries that could use SPI a distinct master, or share the same SPI master.

What is the intended (preferred) way of using a single SPI master when it comes to its initialization and access to its SPI R/W methods?

rsta2, 2020-10-29:
I would do it like that: CSPIMaster is a member of CKernel and will be initialized in `CKernel::Initialize()`. A pointer to this instance and the CE number is handed over to some class constructor in your libraries. With this information you can access the SPI R/W methods from the libraries.

hinxx, 2020-10-29:
Sounds good to me. I later noticed you did exactly that in: 

https://github.com/rsta2/circle/blob/4394f8adff299781e3386482acba2cf91e784ee4/addon/dio/sample/kernel.cpp#L31

Thanks!

closed 2020-10-29 by hinxx