rsta2/circle #153: Adding support for devices

hinxx, 2020-10-31:
I'm working with some SPI / I2C touchscreens and SPI / parallel LCD displays. I would like to contribute the device driver support to upstream.

Would they be accepted, or would they be considered off topic? If yes, which folders should they end up in? Would it be `./lib` or `./addon/xyz`?

rsta2, 2020-10-31:
It depends. If your drivers are for commonly used displays and are not that big (i.e. only a few files) it can be added to addon/something/ or maybe to the existing library in addon/display/. Otherwise, or if you want to apply your own coding style or do not have the time to maintain it on demand, it's better you create your own repository for this project and add it to the library list [here](https://github.com/rsta2/circle/wiki/List-of-Circle-based-projects#libraries).

hinxx, 2020-11-02:
I see you point.
The stuff I would add is not that big, so for that reason alone it could be here.
But maintenance / support (or lack thereof) and coding style make it more attractive for me to do it in my own repo. Thanks!

closed 2020-11-02 by hinxx

rsta2, 2020-11-02:
OK. No problem. Thanks for info.