rsta2/circle #154: Support for USB-serial dongles

hinxx, 2020-11-04:
I've been playing with Ryanteck RTk.GPIO  board and managed to get the basic I/O working. It looks quite useful piece of equipment if one is looking for more I/O (28 pins). I had to introduce rudimentary usbserial.cpp device driver that would talk to CH340 IC.

As a more general support of USB-serial under Circle, I'm also looking at the USB-serial converters that use different popular ICs (CP2102, FT245R, PL2303). As bare minimum support for baud rate, line control (# data bits, # stop bits, parity) and possibly HW flow control would be implemented.

Does it look interesting enough to introduce usbserial to Circle?
In your opinion, how would be best to handle such differences at IC level when introducing USB-serial support? As different IC devices drivers with overarching usbserial layer, or maybe leave the specifics to the app?

rsta2, 2020-11-04:
Yes, USB serial support would be great to have in Circle. Unfortunately I'm not familiar with the differences of the different chips.

Normally I would suggest, to define a unique base class (e.g. *CUSBSerialDevice*), which is derived from *CUSBFunction* and defines the API to the serial interfaces, which should be basically the *Read/Write()* interface of *CDevice* (*CUSBFunction* is already derived from *CDevice*) with some extensions for parameter setting. Currently not sure, how flow control can be realized. The specific chip driver classes can be derived from *CUSBSerialDevice* then and implemented separately. I would try to hide most chip specifics in these specific driver classes, but I don't know, if this is possible with reasonable effort.

There will probably a problem arise with the handling of many different vendor/device ID combinations. I have seen huge lists of this in some Linux driver. I suppose, the current matching method in *usbdevicefactory.cpp* is not suitable for this. We will need an other solution, may be as an extension of the current one.

hinxx, 2020-11-04:
Perfect!

So far, I did the exact same thing that you mention in 2nd paragraph so far in introducing _CUSBSerialDevice_ :). Your suggestion, where specific IC device support would derive _CUSBSerialDevice_, and implement its interfaces, sounds very good to me. I have couple of dongles based on different ICs and will try to figure out how they would support the minimal set of functionality.

Regarding the matching method in _usbdevicefactory.cpp_, it might need some rework to avoid adding numerous VID/PID entries. As it turns out there is no USB class for plain 'serial' device (similar to HID, mass storage, printer) and vendors seem to resort to a combination of `0x00` and `0xFF`values for device and interface class, subclass and protocol values. AFAICT, the VID/PID is the only way to correctly identify such devices.

Could something similar to Linux kernel VID/PID tables work here?
Another approach might be to allow the app to set the VID/PID at instantiation of (IC specific derived) _CUSBSerialDevice_. User would need to know which IC and VID/PID are being used in her/his USB dongle, though.

rsta2, 2020-11-05:
Fine! Thanks for the PR, I will check it. Will need some time. I have already an idea for an USB device ID table (Vendor, Device), which is somewhat similar to Linux. I will implement this.

rsta2, 2020-11-14:
I already did some modifications. I implemented the read buffering, we did spoke about. `Read()` is non-blocking now, which may allow the application to do other things while waiting for data to receive. I had to modify your gpio/rtkgpio class a little bit, because of this. I hope, it still works. My FT232RL dongle is working with your driver.

rsta2, 2020-11-15:
I was facing another issue on the Raspberry Pi 4. Read() reported a timeout there, if the serial USB device did not send data within a second. I applied a fix for this.

I think, these drivers work best in a serial client-server protocol model (client sends request, server responds), because with the missing handshake function, it cannot be used for large amounts of data transferred in one direction. But this shouldn't be a big problem, because this model is frequently used.

rsta2, 2020-11-19:
The USB serial support has been merged to the *develop* branch to be part of the next release. The branch *usb-serial* has been removed.

rsta2, 2020-11-19:
@hinxx I must not forget to thank you again for all your hard work on this feature!

hinxx, 2020-11-20:
@rsta2 don't mention it! I'm happy to be able to contribute to such a great project!

closed 2020-11-20 by hinxx