rsta2/circle #160: USB bulk in transfer length

hinxx, 2020-11-10:
As part of adding the USB to serial support for FTDI FT231X I noticed that the amount of requested bytes on the bulk in pipe, from the device towards host, is not respected by the `CUSBHostController::Transfer()`. For example if I set `nBufSize` to 2, and I use a larger buffer (i.e. 16 bytes) I see that more than 2 bytes have been written into the buffer, but the return value is still set to 2

hinxx, 2020-11-10:
A partial answer is that `CUSBSerialDevice::Read` line https://github.com/rsta2/circle/blob/dd9bff65004af520674d4ff0b130778c746c6209/lib/usb/usbserial.cpp#L143 returns the same value (2 in above example). Maybe it should not. Still does no explain why buffer is written beyond the requested amount.

rsta2, 2020-11-11:
How many more bytes are written to the buffer? The buffer is filled using DMA and always 4 byte values are written. That's one of the reasons, why the `DMA_BUFFER()` macro has to be used to define a buffer, which is used for the USB transfers. Please read [doc/dma-buffer-requirements.txt](https://github.com/rsta2/circle/blob/master/doc/dma-buffer-requirements.txt) for more info on this. Returning just `nCount` in *usbserial.cpp:143* is definitively an error.

rsta2, 2020-11-11:
Something to add. We will probably need a double buffering mechanism in `CUSBSerialDevice::Read()` and `Write()`. Define a buffer with `DMA_BUFFER()` inside these methods and copy the data into this buffer before the write `Transfer()` call or after the read `Transfer()` call. Because the amount of data transferred this way is relatively small, this shouldn't be a problem here.

`Transfer()` returns the number of bytes, which were actually received. This value should be returned by the `Read()` method.

hinxx, 2020-11-11:
Thanks for clarifying the DMA buffer requirements, I failed for read up on that earlier, sorry about that.

> This value should be returned by the Read() method.

Will fix this. I guess the same goes for `Write()` method.

> We will probably need a double buffering mechanism

Will look into that.

> How many more bytes are written to the buffer? 

8 bytes.

Some additional observations.
As it turns out, with FTDI device, host will always get 2 extra bytes prepended to the normal data. In case no data is available only these two bytes are received. Now, in this case, in the buffer I see four bytes being written. The extra two bytes written seem to be a CRC16 value of the first two bytes. With this the `Transfer()` still reports 2 bytes have been received.

For comparison, I have not examined this behavior with other ICs ; which do not add extra to bytes at the start of the data payload.

rsta2, 2020-11-11:
I'm trying to remember the requirements for the USB controller. Because there is no public manual available and I analyzed the Linux driver seven years ago, that's not that easy.

I think, you should not use transfer sizes below MaxPacketSize for `Read()` and the transfer size should be a multiple of MaxPacketSize. On my FT232RL dongle (ven403-6001) the MaxPacketSize is 64 for both endpoints. You can determine the MaxPacketSize using `m_pEndpointIn->GetMaxPacketSize()`.

That would mean, if an application wants to receive less than MaxPacketSize bytes, you still send a transfer request of MaxPacketSize bytes and buffer the remaining data for the next `Read()` call, if you received more data.

The extra data may be the CRC16 used for USB data packets. Perhaps it is written into the buffer by the USB controller and ignored then.

hinxx, 2020-11-11:
I added a check for `Transfer()` minimum size and the DMA aligned double buffers to both, input and output, transfers. With this the extra CRC16 bytes in the receive buffer are not seen anymore. Things look better now.

I did not implement handling of any excess input data for next `Read()` as you mentioned. Partially because I do not have the test in place at the moment, partially because I feel this should be offloaded to the application. Nevertheless, I will try to do something about it and come back to you.

rsta2, 2020-11-12:
I had a quick look on your recent commits and there are things, I did not fully understand. I will have a closer look this afternoon and give you feedback then. My plan is to provide a simple terminal program as a sample. I will be able to test the CP2102 and FT232 support then, if my "ven403-6001" device works with your driver.

rsta2, 2020-11-12:
I had a closer look now. Talking about `CUSBSerialDevice::Read()`: What happens, if the application calls this method with `nCount == 10`, `Transfer()` is called with `count == 64` and the serial device returns 30 bytes for example. `memcpy()` may overwrite some memory then, which does not belong to the application buffer, which is not allowed.

hinxx, 2020-11-12:
You're right, that does not look good. Let me fix that!

Which FT232 variant do you have? It might need a bit different baud rate procedure setting than FT-X I have.

rsta2, 2020-11-12:
Yes, but now you may lose some received characters. They will never be returned to the application. I think, we cannot go here without some kind of FIFO. If you cannot test this, as you wrote, I can only suggest, to take over now, merge your commits and try to improve it. I have most Raspberry Pi models here and the USB serial support has to be tested on them all in any case before release.

My FT232 dongle is a FT232RL with vendor ID 0x403 and device ID 0x6001.

hinxx, 2020-11-12:
I understand the issue of possible data loss, it might be best for you to take over from here. Thank you!

As far as FT232RL goes, baud rate setting should be fine as-is, you might want to add it as a recognized device type somewhere here: https://github.com/hinxx/circle/blob/b72726495a8672867aa81eca1183aa3dedce634c/lib/usb/usbserialft231x.cpp#L66

rsta2, 2020-11-12:
OK, I will take over. Thank you for all your work so far! I will add the FT232RL device and test it. I will let you know the results.

closed 2020-11-12 by hinxx