rsta2/circle #162: German keyboard - hash key has no function

olfp, 2020-11-20:
The hash key does not work on a german keyboard due to a wrong scan code.
keymap_de.h has the # and ' codes for scan code 0x31, correct is 0x32 (where the map has no codes).
Swapping lines for 0x31 and 0x32 (and correction the comments) resolves the bug. 
Tested on Rpi 3B+ with circle 43.1

rsta2, 2020-11-20:
There seem to be different hash key mappings. My old keyboard has it at 0x31, now I tested a newer one, and indeed the hash key is at 0x32 there. Because this key position was free, I simply added the same mapping to 0x32. The update is on the *develop* branch at the moment and will be part of the next release. Thanks for reporting this!

rsta2, 2020-11-24:
The fix is part of the new release Step43.2. This can be closed.

closed 2020-11-24 by rsta2