rsta2/circle #164: With DHC activated and no network (i.e. cable unpluged) startup hangs forever

olfp, 2020-11-23:
When configuring a kernel to use DHCP and then not having a network connection (cable unplugged) leaves the system waiting for DHCP forever (four tries, error message, another four, repeat ...)
For testing I exposed m_bUseDHCP as public in include/circle/net/netsubsystem.h and modified lib/net/dhcpclient.cpp as follows:
                case DHCPStatusConfigChanged:
                        //CScheduler::Get ()->Sleep (RESTART_DELAY);
                        //goto InitState;
                        m_pNetSubSystem->Get ()->m_bUseDHCP = 0;
                        return;
This way the system continues and executes my user code. 

I think it would be nice if there was a standard way to configure DHCP non-blocking / graceful failing.

dwhinham, 2020-11-23:
If you look at `CNetSubSystem::Initialize()`, it takes a boolean `bWaitForActivate` which defaults to `TRUE`: https://github.com/rsta2/circle/blob/4394f8adff299781e3386482acba2cf91e784ee4/include/circle/net/netsubsystem.h#L46

If you pass `FALSE`, `Initialize()` will return immediately. DHCP will then continue to retry in a background task, and you can check the return value of `CNetSubSystem::IsRunning()` later, which will be `TRUE` when DHCP is ready.

Does passing `FALSE` not achieve what you are looking for?

olfp, 2020-11-23:
In addition I just tested with sample 18-ntptime with (Default) DHP enabled. This also initialises networking  with CNetSubSystem::Initialize(false). 
Here also the first line in Kernel::Run logging the kernel name and compile time never runs but the system keeps waiting for DHCP.

hpingel, 2020-11-23:
If Initialize returns false in your application based on CStdlibAppNetwork, you would need to give the scheduler a chance to work on other tasks with m_pScheduler->Yield () in your application I guess.

hpingel, 2020-11-23:
In example 18-ntptime CKernel::Initialize will return FALSE if no network cable was attached. This will cause the system to halt due to the logic in main.cpp which explains why you don't see the logger line. Remove line 30+31 in main.cpp and test again.

https://github.com/rsta2/circle/blob/master/sample/18-ntptime/kernel.cpp#L98
https://github.com/rsta2/circle/blob/master/sample/18-ntptime/main.cpp#L30

olfp, 2020-11-23:
Re-tested example 18-ntptime  with lines 30+31 in main.cpp commented out. No change.
Forever tries to get an DHCP address.

hpingel, 2020-11-23:
Well, I have an application running using circle-stdlib 15.3 and I can confirm that this application is able to deal fine with a missing network cable. But I don't use CStdlibAppNetwork because only want to initialize net when the user triggers it during runtime.

Could you post a photo of your HDMI screen output?

rsta2, 2020-11-23:
@olfp Perhaps you have overseen the "Compile time" message in CKernel::Run() in sample/18-ntptime? I tested the sample again on a RPi 3B+ and the message comes (very early) in the third line. Then the sample waits in a loop until `m_Net.IsRunning ()` becomes true. Of course the sample "hangs" there, because it needs the network and thus an IP address from DHCP. If you adapt this sample, you can do whatever you want in this loop, but not accessing the network, of course.

@smuehlst It's right, that the circle-stdlib project does not support this yet. There should be a boolean parameter `bWaitForActivate` for CStdlibAppNetwork::Initialize(), which defaults to true and the final loop in this method should be jumped over, if the parameter is false. Perhaps this can be added.

olfp, 2020-11-23:
@rsta2: You are right, I oversaw the "Compile time" message (it scrolled out, only 25 lines on the screen).
I now removed any network related code from my kernel.cpp. So there is only the DHCP initialisation via CStdlibAppNetwork.
The "Compile time" message i present but the following printf() never shows up on the screen instead I just get the DHCP loop: Four "No response from server" messages, one "Did not receive OFFER", repeat.
I also modified 18-ntptime to output something in the IsRunning() loop, and yes, it works.
Thank you for the help, I will investigate further.

smuehlst, 2020-11-23:
> @smuehlst It's right, that the circle-stdlib project does not support this yet. There should be a boolean parameter bWaitForActivate for CStdlibAppNetwork::Initialize(), which defaults to true and [...]

@rsta2 Sure, that can easily be added. I will take a note, and when you create a new Circle release, I will add that to the corresponding circle-stdlib release.

olfp, 2020-11-24:
I nailed it. The issue is not with circle but with circle-sodlib, more specific CStdlibAppNetwork. Thank @rsta2 for pointing out the wait loop in 18-ntptime, that put me on the track!
In circle-stdlib/include/circle_stdlib_app.h fro like 28 on there is the loop  
                while (!mNet.IsRunning ())
                {
                        mScheduler.Yield ();
                }
which turns the non waiting initialisation via CStdlibAppNetwork::Initialize(false) into a waiting one. Adding a timeout to that loop correctly starts the kernel (and if I plug in the network later, it receives an IP address):
                unsigned nTime = mTimer.GetTime();      
                while (!mNet.IsRunning () &&
                        ((mTimer.GetTime() - nTime) < 10))
                {
                        mScheduler.Yield ();
                }
@smuehlst: If changing circle_stdlib_app.h like this is OK, made you can add it (perhaps with a configurable wait time) when you make the change for bWaitForActivate?
@rsta2: Unrelatedly, how about assigning a self-assigned I Address (169.254.x.x) when DHCP does not receive an offer?

closed 2020-11-24 by olfp

rsta2, 2020-11-24:
@olfp We have planned to completely eliminate this `while (!mNet.IsRunning())` loop, if the new boolean parameter is false in `CStdlibAppNetwork::Initialize()`. With this it's completely up to the application code in `CKernel::Run()` to handle DHCP failed conditions.