rsta2/circle #165: CTimer crashes if CLogger is not setup

msx80, 2020-11-24:
Hi there! I ran into this issue: CTimer was crashing my app on Initialize(). After some debugging i found the culprit: in a couple of methods in timer.cpp, some logs are written using `CLogger::Get ()->Write`. If logger is not initialized, Get() returns null and the Write fails. I don't require instantiating a logger in this particular program (nor a Screen or other utility stuff), and it shouldn't be required for a Timer to be used since it works perfectly without.

You could either remove the logs, or testing for Get() before calling Write, or create some static CLogger::Write() that does the check internally and eventually discard the log.

PS the same problem could be present in other classes too.

Thanks in advance!

rsta2, 2020-11-25:
Yes, only relatively simple programs can work without an instance of CLogger. In this specific case you can undefine `CALIBRATE_DELAY` in *include/circle/sysconfig.h* or add `DEFINE += -DNO_CALIBRATE_DELAY` to *Config.mk* so that CTimer does not call `CLogger::Write()` any more.

But yes, CLogger is used throughout the system to report events and information, which can be interesting for the developer or user. There is a simple method to get rid of the logger messages, if you do not need them. Just create an instance of `CNullDevice` (see <circle/nulldevice.h>) and initialize the logger with this null device as the target.

msx80, 2020-11-25:
You're right that a logger is mostly wanted, i still feel like it's a bit ugly to have such hidden dependencies. But i can live with that, i'll use the NullDevice :) Thanks!

closed 2020-11-25 by msx80

rsta2, 2020-11-26:
I found a simple but better solution for this problem. `CLogger::Get()` is creating a dummy instance of `CLogger`, if this has not been done before. With this the developer does not need to care about the logger any more, if he is not interested in the messages. The update is currently on the *develop* branch. Thanks for pointing me on this!