rsta2/circle #166: Pi 400: "xhci: HW reset failed" on boot

olfp, 2020-11-26:
While support for the Pi 400 is mentioned nowhere, I decided to give it a try as it's supposed to be a Pi 4 logically.
Unfortunately I get "xhci; HW reset failed" as the fourth log line on the HDMI screen when trying to run my favourite  
sample, 32-i2cshell. Then it hangs.
Screenshot attached, the message appears after logger, timer and pcie as you can see.

Is  the Pi 400 supported? If not, are there plans to do so?

Best regards
Olaf

[image]

rsta2, 2020-11-26:
I didn't know, if the Raspberry Pi 400 would work with Circle, and this is noted in the first paragraph of the main README.md file. So it seems, at least the USB is not working. I do not have a RPi 400 and currently I do not plan to buy one. So it's difficult. We could try to sort this out using Email, but it will probably take time. I may send you patches and you may send the results. I don't know, if you want to take part on this? BTW. Do you have updated the firmware version to the one downloaded with `make` in boot/?

olfp, 2020-11-26:
Ooops, in the first paragraph...
I did „make“ in boot on a fresh cloned circle instance, made armstub7-rpi4.bin, copied everthing marked „Pi 4“ and copied config32.txt as config.txt.
If you need an RPi 400, I could borrow mine, it‘s not used now.

rsta2, 2020-11-26:
Thanks for testing and for your offer. I did some reading and how it seems, this may be a general problem also for other newer RPi 4 model revisions with at least 4 GB RAM. I first have to understand, what can be done about this. Stay tuned!

rsta2, 2020-11-30:
@olfp I suppose, I have to implement a parser for the devicetree blob, loaded by the firmware, to generally solve this problem. Because it seems, there is also some firmware development still ongoing in this field, it should be OK when the RPi 400 support will be ready in a few weeks. How I heard, until then I will also get a RPi 400 from somewhere else, so I can test it. So we do not need to send your RPi 400 around. Thanks again.

olfp, 2020-12-04:
Works like a charm on the Pi 400! After downloading the "develop" branch I could run the sample out of the box.
After fiddling in the changes with my current project tree I can now run my project on the 400 too.
Thank you very much!

closed 2020-12-04 by olfp